bluerobotics/BlueOS #1720: [wizard] Retry not working

joaoantoniocardoso, 2023-05-26:
https://github.com/bluerobotics/BlueOS-docker/assets/5920286/a3494188-a160-4496-b857-ab6e9e23ce32

RaulTrombin, 2023-06-02:
Same here, testing on:
Fresh install BlueOS Version 1.0.1

[image]
Retry not working.

This one with Reset Settings and reboot.
[image]

But here it works:
[image]

Williangalvani, 2023-06-09:
the current wizard executes the promises in the wrong step, which means they run as soon as they are created, and the "retry" button just recreates the list with the already resolved promises

PR #1760 (2023-06-12): Allow embedding parameters into firmware
supports injecting them into the .apj firmware for stm32 boards, and creating a .params file in userdata for linux

https://github.com/bluerobotics/BlueOS-docker/assets/4013804/2a471371-c7af-4c92-9682-41fe34638f10

ps: I just increased the width of the select so the full label shows up

additional fix for #1734

this can be tested by running the wizard and chosing a default set, and then resetting the parameters to firmware defaults in qgc and checking if the configuration remains (easier to see frame type in the 3d viewer, for sub at least)

fix #1720  [... cut]
closed 2023-06-12 by Williangalvani