bluerobotics/BlueOS #1722: [wizard] Error: Failed to install firmware: Error: timeout of 30000ms exceeded.

joaoantoniocardoso, 2023-05-26:
The wizard is failing to download the firmware every time here.

It also shows `Could not set hostname: Request failed with status code 500. ` on the bottom as a notification.

[image]

patrickelectric, 2023-05-26:
can you provide the steps to replicate ?

joaoantoniocardoso, 2023-05-26:
1. Reset settings
2. follow the wizard

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/a5ca00bf-f3e3-4b55-a2bb-6db188192249

joaoantoniocardoso, 2023-05-26:
Note that I'm using Raspberry Pi 3 and Pixhawk, on latest master

Williangalvani, 2023-06-09:
this is also related to #1720. the promises are running before we have internet, I think.

PR #1760 (2023-06-12): Allow embedding parameters into firmware
supports injecting them into the .apj firmware for stm32 boards, and creating a .params file in userdata for linux

https://github.com/bluerobotics/BlueOS-docker/assets/4013804/2a471371-c7af-4c92-9682-41fe34638f10

ps: I just increased the width of the select so the full label shows up

additional fix for #1734

this can be tested by running the wizard and chosing a default set, and then resetting the parameters to firmware defaults in qgc and checking if the configuration remains (easier to see frame type in the 3d viewer, for sub at least)

fix #1720  [... cut]
closed 2023-06-12 by Williangalvani