bluerobotics/BlueOS #1724: [extension manager] [suggestion] Set all selection checkbox to ON by default to allow easy inverse filtering

joaoantoniocardoso, 2023-05-26:
Now the default is all set to off:

[image]

But if we change to all set to on, it would allow for easy inverse filtering, like to hide examples with 1 click:
[image]

patrickelectric, 2023-05-28:
That's not the default behavior of filters on the websites. If you go to Google shopping, Amazon and others. The filters are not selected.

joaoantoniocardoso, 2023-05-29:
> That's not the default behavior of filters on the websites. If you go to Google shopping, Amazon and others. The filters are not selected.

I know, also, all selected makes many more clicks to select only one category.

Probably we just need to split it into two major categories -- actual extensions and examples

ES-Alexander, 2023-06-26:
I think the problem you're describing would be best solved by "select all"/"clear all" selectors, which are a reasonably common UI element.

That said, I think the problem you're actually trying to solve here would be better solved by implicitly sorting the extensions by type to start with, whereby `device-integrations` always come first, with `other` and then `examples` last, which just seems like a reasonable thing to do since the types are fixed and have a reasonably logical importance order already. We're also [already doing that on the online store display](https://docs.bluerobotics.com/BlueOS-Extensions-Repository/).

closed 2023-10-11 by joaoantoniocardoso