bluerobotics/BlueOS #1731: [wifi-manager] Confusing UI behavior

joaoantoniocardoso, 2023-05-26:
Two weird behavior can be seen in the video below:
1. password was asked again after just disconnecting from the network
2. 'forget' password button keeps showing even after we clicked 'forget'

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/ba972892-08e6-4fc3-ba6f-dff6ee8442a0

joaoantoniocardoso, 2023-06-09:
Just updating, here I'm showing that sometimes the wifi manager refuses to forget the password:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/c4176c5a-7884-419a-b4c5-9babcd26813d

joaoantoniocardoso, 2023-08-04:
Also reported by Eldin using beta 27. @patrickelectric

patrickelectric, 2023-08-04:
On master ?

patrickelectric, 2023-08-04:
Ok, I'll try to talk with him and get what is going on.

patrickelectric, 2023-08-04:
It should be fixed by https://github.com/bluerobotics/BlueOS/pull/1899

joaoantoniocardoso, 2023-08-04:
> It should be fixed by #1899

But he was using beta 27, which contains that commit

patrickelectric, 2023-08-10:
Waiting for Eldin to confirm the issue and how to replicate it.

patrickelectric, 2025-01-16:
Tested with latest master

closed 2025-01-16 by patrickelectric