bluerobotics/BlueOS #1742: Duplicated autopilot parameters

rafaellehmkuhl, 2023-06-03:
[image]

I was using SITL and changed to Navigator. Don't know if this was the reason, though.

Williangalvani, 2023-06-06:
interesting. we should probably empty the store when changing fw, though that doesn't quite explain the issue.. 🤔

rafaellehmkuhl, 2023-06-06:
I think that explains the issue, no?

Williangalvani, 2023-06-06:
> I think that explains the issue, no?

yeah, I just checked and the keys in ParametersTable are numbers, which allows for duplicated names.

joaoantoniocardoso, 2023-06-07:
I couldn't reproduce it by changing from SITL to Pixhawk, and back to SITL again.

rafaellehmkuhl, 2023-06-07:
> I couldn't reproduce it by changing from SITL to Pixhawk, and back to SITL again.

Strange...
I changed from SITL to Navigator, but unless there's some contamination between the Linux routines, this shouldn't make a difference.

PR #1756 (2023-06-08): core: frontend: reset frontend parameters when flashing new firmware and restarting
should fix #1742