bluerobotics/BlueOS #1757: Cameras freezing on RPI 3

joaoantoniocardoso, 2023-06-07:
On Raspberry Pi 3, USB camera streams (using video4linux) are constantly freezing in a non-periodic interval, as shown below:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/878eda94-1a8b-483b-98f9-69313059e76d

Facts:
- freezes on master
- freezes on stable
- doesn't freeze on the old companion
- GStreamer version doesn't matter (tested with 1.8, 1.18, 1.20, and 1.22)

More details on the [original post (internal slack) ](https://bluerobotics.slack.com/archives/C015VGZT5K6/p1685955211923179)

joaoantoniocardoso, 2023-06-07:
After many tests, the villain was found to be the `dwc2` overlay, used to add OTG capability to the Raspberry Pi 4.

With that said, we'd need to disable this overlay for RPI3.

RaulTrombin, 2023-06-08:
So, I think it depends on where the Pi image is generated:

1. In the [test-and-deploy.yml](https://github.com/bluerobotics/BlueOS-docker/blob/master/.github/workflows/test-and-deploy.yml) file step generates the image on a self-hosted environment, which should be Raspberry Pi, likely a Raspberry Pi 

2. In the [blueos.Pifile](https://github.com/bluerobotics/BlueOS-docker/blob/master/deploy/pimod/blueos.Pifile), the pimod calls the install.sh script.

3. In the [install.sh](https://github.com/bluerobotics/BlueOS-docker/blob/master/install/install.sh), the default setting for DO_BOARD_CONFIG is true, so it calls the [config_board.sh ](https://github.com/bluerobotics/BlueOS-docker/blob/master/install/boards/configure_board.sh) script.

4. In the [boards directory](https://github.com/bluerobotics/BlueOS-docker/tree/master/install/boards), the bc28 folder seems to have fewer overlays for the Raspberry Pi 3, but it is not actually used because the setup is running on a Raspberry Pi 4.

We can have the board config on Pfile with the install.sh, but do not execute it when on CI,

Then try this method, which will be called on first boot on the raspberry target:
https://elinux.org/RPi_raspi-config#First-boot_activity
https://github.com/nmcclain/raspberian-firstboot

joaoantoniocardoso, 2023-06-08:
For the record, this bug will happen to any RPI3 that had updated at least once its BlueOS to any version since Apr. 24 (between 1.1.0-beta.19 to the current master or 1.1.0-beta.22), or if someone added the dwc2 overlay for any reason.

PR #1765 (2023-06-20): Do not apply dwc2 OTG overlay to RPI3
Fixes #1757 

- [x] Fixes running systems
  - [x] Fixes OTG messing with RPI3
  - [x] Moves all Navigator-related configurations to pi4 section
  - [x] Don't mess with user configurations (as long as they place an inline comment with the word `custom` on it)
  - [x] Backup files before patch them
- [ ] ~~Fixes new images~~

We should document the use of the 'custom' inline comment to protect the line from this script