bluerobotics/BlueOS #1770: wizard/wifi/network: 'waiting for internet' even after connected to wifi

joaoantoniocardoso, 2023-06-09:
This is probably caused by the default route still being the ethernet (cabled) device:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/3beff309-5ba3-410b-943e-8768b343cb8f

Williangalvani, 2023-06-10:
it is actually checking for internet connectivity to some websites, so wifi itself may not be enough if you have a weird setup:

https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/helper/main.py#L155-L158

patrickelectric, 2023-06-21:
Related to https://github.com/bluerobotics/BlueOS-docker/issues/1539

patrickelectric, 2023-06-28:
@joaoantoniocardoso is that a thing ?

joaoantoniocardoso, 2023-06-28:
> @joaoantoniocardoso is that a thing ?

I'm not sure whether changing the priority solves it, it seems to be related to ipv4 vs ipv6. In the image below, the output of the first `route` is with the cable connected but no internet. Then I kept the cable, removed the internet connection from that ethernet cable, and connected to the WiFi network, which is connected to the internet, but somehow only ipv6 works from inside BlueOS:

[image]

With my laptop connected to the same WiFi I can `ping -4` with no problems when using that same WiFi.

joaoantoniocardoso, 2023-06-28:
More info:

[image]