bluerobotics/BlueOS #1796: core: frontend: Vehicle Setup: add sensor calibration / configuration

ES-Alexander, 2023-06-15:
### Current behaviour

We have PWM output configuration (including basic motor testing), and we display detected sensors but don't allow calibrating or configuring them. Accordingly the only way to configure sensors directly through BlueOS is to know about and modify relevant parameters by hand, which is not a pleasant or intuitive user experience.

### Expected or desired behaviour

There's decent scope to make an intuitive interface for sensor setup, and doing so could help highlight several vehicle functionalities that exist but are currently under-utilised because they're not known about.

This has been the intent from the start, but hasn't been completed yet, so I'm adding this issue to track progress on it.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2023-06-15:
Per [internal discussion](https://bluerobotics.slack.com/archives/CB4P9U40H/p1686788553414979?thread_ts=1686686754.821739&cid=CB4P9U40H) it would be helpful to include nice configuration for things like [compass declination](https://docs.bluerobotics.com/ardusub-zola/software/autopilot/ArduSub-4.1/developers/parameters/#compass-dec-compass-declination), particularly since the units for the raw parameter are in radians (which generally requires some form of calculation software to determine values for).

Williangalvani, 2023-10-30:
Sensors:
 - [ ] Gyro Calibration
 - [ ] Baro zero-depth calibration
 - Accelerometer
     - [ ] full calibration
     - [ ] dance-less calibration
 - Compass
     - [ ] full dance and read feedback
     - [ ] fast-one

patrickelectric, 2024-07-11:
@Williangalvani this is done right ?

Williangalvani, 2024-07-11:
> @Williangalvani this is done right ?

we are missing the gyro calibration and level horizon parts

ES-Alexander, 2025-02-17:
It would also be nice to include an intuitive interface for specifying flight controller board orientation.

ES-Alexander, 2025-02-18:
Similarly for `BARO_ALT_OFFSET`, and ArduSub's `BARO_SPEC_GRAV` and `SURFACE_DEPTH` parameters.

[Relevant post for context](https://discuss.bluerobotics.com/t/accessing-depth-information/19401/4).

patrickelectric, 2026-09-11:
Already available in the Configure tab since #2342 and #2451: compass, accelerometer, gyro, baro, power, camera and failsafe.

closed 2026-09-11 by patrickelectric