bluerobotics/BlueOS #1799: [Wizard] During firmware installation, we should automatically reboot autopilot before retrying to install the firmware

joaoantoniocardoso, 2023-06-16:
We should automatically reboot autopilot during firmware installation before retrying to install the firmware, at least for Pixhawk. I tried the 'retry' three times before I decided to record the following video.

Also, note that I had to reload the page to get out of that situation.

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/1d5a3d94-f636-4c4a-a080-f68e7f112e38

ES-Alexander, 2023-06-22:
Perhaps worth noting that if Pixhawk flashing fails (which should be rare) it generally requires a full power reboot before the bootloader becomes available.

That said, most issues recently seem to be with finding and fetching the firmware from the ArduPilot servers, which is less of a problem. I'm not sure if/why that would require restarting the autopilot, but I suppose that could help if it gets put in some weird state when the firmware fetch process is started or something 🤷‍♂️

patrickelectric, 2023-07-04:
Can you provide the steps to replicate ? @joaoantoniocardoso I did try some crazy things here and it end up working without problems for edge cases.

joaoantoniocardoso, 2023-07-05:
> Can you provide the steps to replicate ? @joaoantoniocardoso I did try some crazy things here and it end up working without problems for edge cases.

1. Flash your Pixhawk with a random image, e.g.: the stable copter
2. Enable the wizard, it will fail to flash, no matter how many times you try.
3. Skio/cancel the wizard, go to the ardupilot firmware page, restart the Pixhawk.
4. Then, refresh BlueOS to see the wizard again, proceed with the steps and it will succeed.

joaoantoniocardoso, 2023-07-13:
[image]

It seems that it is a front-end problem: it is writing the firmware successfully, but it says otherwise:

[image]

rafaellehmkuhl, 2023-07-13:
> [image]
> 
> It seems that it is a front-end problem: it is writing the firmware successfully, but it says otherwise:
> 
> [image]

It can be that the firmware is being successfully installed, but it took more time than the timeout for the HTTP request. Worth checking.

patrickelectric, 2023-08-11:
@joaoantoniocardoso to fix this issue, used the reboot function on the frontend, that for usb flight controllers is limited to:

```
message = self.command_long_message("MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN", [1.0])
await self.mavlink2rest.send_mavlink_message(message)
```

PR #1979 (2023-08-11): core: frontend: Wizard: Reboot flight controller if it fails to flash on the first try
Fix #1799
closed 2023-08-11 by patrickelectric