bluerobotics/BlueOS #1801: bug: Beta 23, unable to load navigator board

RaulTrombin, 2023-06-16:
### Bug description

[image]

A fresh install was made with beta 23, it boots without a navigator HAT attached,
On the next boots the user is unable to change the board.

Even changing to master, reset settings, etc.

### Steps to reproduce

fresh install a image and boot without navigator.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2023-06-16:
it seems to be failing to detect the board. the the autopilotmanager logs

RaulTrombin, 2023-06-22:
we can close this, it's an issue related only to the previous navigator board.

closed 2023-06-22 by RaulTrombin