bluerobotics/BlueOS #1804: Motor tests when in Auto mode are dangerous

rafaellehmkuhl, 2023-06-16:
We should put the vehicle in Manual mode before starting the test, otherwise if there's a mission in the vehicle, the motors will spin.

From @raimullin14:

"Just a heads up. I was disarmed in auto mode on cockpit and when I armed it in this window it resumed the mission in cockpit and started running the thrusters".

[image]

patrickelectric, 2023-06-21:
It looks much more as a firmware problem than from our side, the firmware should not allow motor test in any mode besides MANUAL. @Williangalvani

ES-Alexander, 2023-06-21:
It's a firmware _problem_, but given it's already the case in the current firmware version(s) it seems like it wouldn't hurt for BlueOS to just send a manual mode change before trying to do motor test stuff.

Williangalvani, 2023-06-22:
Alternatively, we can disable motor test in modes other than manual

patrickelectric, 2023-06-22:
Yes that's required, but a palliative solution for the real problem.

PR #1850 (2023-07-05): Require manual mode for motor test
Closes #1804
closed 2023-07-05 by patrickelectric