bluerobotics/BlueOS #1807: Rebooting the core conntainer is also rebooting the raspberry

joaoantoniocardoso, 2023-06-20:
Possibly related to PR #1765, the current master reboots the Linux after the container when we ask it to reboot the container:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/c005323d-e004-4eaf-b60d-a13dd58ca668

joaoantoniocardoso, 2023-06-20:
Well, it seems that it isn't related to the possible `reboot` added by #1765, otherwise, it would show a message, and wait for 10 seconds before rebooting it.

Also, it doesn't always happen, as shown in the video below:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/b9cb5b58-9808-44f7-a4e5-3e0149a47ae6

joaoantoniocardoso, 2023-06-20:
@Williangalvani any idea?

Williangalvani, 2023-06-21:
> @Williangalvani any idea? Might be related to bootstrap..?

I don't think so. I don't think it would reboot, much more less likely with no warning.

what are the first lines of `journalctl -r -b 1` after a reboot?

joaoantoniocardoso, 2023-06-21:
> > @Williangalvani any idea? Might be related to bootstrap..?
> 
> I don't think so. I don't think it would reboot, much more less likely with no warning.
> 
> what are the first lines of `journalctl -r -b 1` after a reboot?

I'm not really sure what's going on with the dates here, but:
```
-- Journal begins at Fri 2022-01-28 01:22:52 GMT, ends at Wed 2023-06-21 03:48:19 BST. --
Jun 09 20:37:18 blueos systemd-journald[128]: Journal stopped
Jun 09 20:37:18 blueos systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Jun 09 20:37:17 blueos systemd-shutdown[1]: Syncing filesystems and block devices.
Jun 09 20:37:17 blueos systemd[1]: Failed to set timeout to 600s: Invalid argument
Jun 09 20:37:17 blueos systemd[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog
Jun 09 20:37:17 blueos systemd[1]: Shutting down.
Jun 09 20:37:17 blueos systemd[1]: Reached target Reboot.
Jun 09 20:37:17 blueos systemd[1]: Finished Reboot.
Jun 09 20:37:17 blueos systemd[1]: systemd-reboot.service: Succeeded.
Jun 09 20:37:17 blueos systemd[1]: Reached target Final Step.
Jun 09 20:37:17 blueos systemd[1]: Reached target Shutdown.
```

for `journalctl -r -b`:

```
-- Journal begins at Fri 2022-01-28 01:22:52 GMT, ends at Wed 2023-06-21 03:52:56 BST. --
Jun 21 03:52:56 blueos sshd[1748]: banner exchange: Connection from 127.0.0.1 port 41124: invalid format
Jun 21 03:52:56 blueos sshd[1748]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:51:47 blueos sshd[1742]: banner exchange: Connection from 127.0.0.1 port 40162: invalid format
Jun 21 03:51:47 blueos sshd[1742]: error: send_error: write: Broken pipe
Jun 21 03:51:47 blueos sshd[1742]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:50:39 blueos sshd[1735]: banner exchange: Connection from 127.0.0.1 port 39194: invalid format
Jun 21 03:50:39 blueos sshd[1735]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:49:30 blueos sshd[1727]: banner exchange: Connection from 127.0.0.1 port 38206: invalid format
Jun 21 03:49:30 blueos sshd[1727]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:48:19 blueos sshd[1723]: banner exchange: Connection from 127.0.0.1 port 37208: invalid format
Jun 21 03:48:19 blueos sshd[1723]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:47:31 blueos dhcpcd[699]: eth0: part of Router Advertisement expired
Jun 21 03:47:31 blueos dhcpcd[699]: eth0: fe80::dac6:78ff:fe61:e760: router expired
Jun 21 03:47:09 blueos sshd[1714]: banner exchange: Connection from 127.0.0.1 port 36136: invalid format
Jun 21 03:47:09 blueos sshd[1714]: error: send_error: write: Broken pipe
Jun 21 03:47:09 blueos sshd[1714]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:46:02 blueos sshd[1712]: banner exchange: Connection from 127.0.0.1 port 35124: invalid format
Jun 21 03:46:02 blueos sshd[1712]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:45:00 blueos sshd[1668]: banner exchange: Connection from 127.0.0.1 port 33762: invalid format
Jun 21 03:45:00 blueos sshd[1668]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:45:00 blueos sshd[1667]: banner exchange: Connection from 127.0.0.1 port 33760: invalid format
Jun 21 03:45:00 blueos sshd[1667]: error: kex_exchange_identification: client sent invalid protocol identifier "GET / HTTP/1.1"
Jun 21 03:44:53 blueos systemd-timesyncd[1622]: Initial synchronization to time server 200.160.7.186:123 (0.debian.pool.ntp.org).
Jun 21 03:44:53 blueos systemd[1]: Started Network Time Synchronization.
Jun 21 03:44:53 blueos systemd[1]: Starting Network Time Synchronization...
Jun 21 03:44:52 blueos systemd[1]: Stopped Network Time Synchronization.
Jun 21 03:44:52 blueos systemd[1]: systemd-timesyncd.service: Succeeded.
Jun 21 03:44:52 blueos systemd[1]: Stopping Network Time Synchronization...
Jun 21 03:44:52 blueos avahi-daemon[372]: Failed to parse address 'fe80::dac6:78ff:fe61:e760%eth0', ignoring.
Jun 21 03:44:52 blueos avahi-daemon[372]: Got SIGHUP, reloading.
Jun 21 03:44:52 blueos avahi-daemon[372]: Registering new address record for 192.168.0.87 on eth0.IPv4.
Jun 21 03:44:52 blueos dhcpcd[699]: eth0: adding default route via 192.168.0.1
Jun 21 03:44:52 blueos dhcpcd[699]: eth0: adding route to 192.168.0.0/24
Jun 21 03:44:52 blueos dhcpcd[699]: eth0: leased 192.168.0.87 for 14400 seconds
Jun 21 03:44:48 blueos kernel: ---[ end trace 641ece6062815be6 ]---
Jun 21 03:44:48 blueos kernel: 7fe0: 00000003 76daf898 76f395b9 76f3b676
Jun 21 03:44:48 blueos kernel: 7fc0: 00001000 76a00ad8 76db0590 00000003 00000000 00001000 00001000 00000000
Jun 21 03:44:48 blueos kernel: 7fa0:                   00001000 76a00ad8 00000020 76a00ad8 00001000 00000000
Jun 21 03:44:48 blueos kernel: Exception stack(0x96f67fa8 to 0x96f67ff0)
Jun 21 03:44:48 blueos kernel: [<80332edc>] (sys_read) from [<80100040>] (ret_fast_syscall+0x0/0x28)
Jun 21 03:44:48 blueos kernel:  r4:00001000
Jun 21 03:44:48 blueos kernel:  r10:00000003 r9:96f66000 r8:80100204 r7:00000003 r6:76db0590 r5:76a00ad8
Jun 21 03:44:48 blueos kernel: [<80332dfc>] (ksys_read) from [<80332ef4>] (sys_read+0x18/0x1c)
[... cut ...]
Jun 21 03:44:25 blueos systemd[1]: systemd-hostnamed.service: Succeeded.
Jun 21 03:44:25 blueos systemd-logind[397]: Removed session 8.
Jun 21 03:44:25 blueos systemd-logind[397]: Session 8 logged out. Waiting for processes to exit.
Jun 21 03:44:25 blueos systemd[1]: session-8.scope: Succeeded.
Jun 21 03:44:25 blueos sshd[1088]: pam_unix(sshd:session): session closed for user pi
Jun 21 03:44:25 blueos sshd[1094]: Disconnected from user pi ::1 port 59770
Jun 21 03:44:25 blueos sshd[1094]: Received disconnect from ::1 port 59770:11: disconnected by user
```

patrickelectric, 2023-06-21:
Unable to replicate doing normal "soft reboot" via BlueOS interface. 
I also did try `docker stop blueos-core`, but the raspberry did not reboot, BlueOS restarted without problems.

Tested in beta.23

joaoantoniocardoso, 2023-06-22:
> Unable to replicate doing normal "soft reboot" via BlueOS interface. I also did try `docker stop blueos-core`, but the raspberry did not reboot, BlueOS restarted without problems.
> 
> Tested in beta.23

It is happening on master only

patrickelectric, 2023-06-22:
Can you double check if it's `master` only ? Otherwise it'll be the startup script, and that's weird..

[image]

joaoantoniocardoso, 2023-06-22:
> Can you double check if it's `master` only ? Otherwise it'll be the startup script, and that's weird..
> 
> [image]

I tried to reproduce it in beta 23 several times, seems to be reproducible only on master, and **it doesn't always happen**, so there's a chance for it to be a combination of factors.

joaoantoniocardoso, 2023-06-22:
Also, there's a chance that this only happens on Pi3, I need someone to reproduce it on a Pi4

patrickelectric, 2023-06-22:
tested on master and in a pi4, no problems.

joaoantoniocardoso, 2023-07-12:
Just to mention, it's happening on pi4 too

patrickelectric, 2023-07-12:
Is weird how that got worst on beta.26, the changes on the code appears to not match the pattern. That may help somehow.

PR #1865 (2023-07-13): Fix core: tools: blueos_startup_update: Make the order of the configs deterministic
lesson learned: the order of the elements of a set is not deterministic.

Closes #1807
closed 2023-07-13 by patrickelectric