bluerobotics/BlueOS #1808: CPU usage profile during Blueos initialization

joaoantoniocardoso, 2023-06-20:
This isn't an issue, it is just to show our CPU usage profile during initialization.

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/45500e25-df8e-431f-820e-425dad7eeb34

patrickelectric, 2023-06-21:
From what I could detect during a previous personal investigation, is that starting the http servers uses a huge amount of cpu. We should consider improving it somehow, maybe cashing something on python compiler side ? Don't know.

patrickelectric, 2023-06-21:
I did a quick test with nuitka, I was able to get it working and with others services, but it appears to compile only the python code that it can see, without doing some smart stuff that pypy could.

Talking about pypy, they do not provide binaries for arm32, only arm64. It may worth to add it in our base image and checkout.