bluerobotics/BlueOS #1821: CI: create Boat and Rover ready images for bluerov2 and BB120

Williangalvani, 2023-06-22:
they could just take the .img file and add the additional preset files to it.

patrickelectric, 2024-06-19:
@Williangalvani this is done, right ?

ES-Alexander, 2025-05-05:
This is currently worked around by the setup wizard including a "select your vehicle type" step, but results in a "the software should know this - why do I need to specify it?" situation for users of stock vehicles, which is frustrating and can needlessly lead to incorrect setups that are annoying to recover from.

From internal discussion:

> @ES-Alexander: Given our BlueROV2 and BlueBoat have different hardware, could we reconfigure the BlueOS startup wizard to detect that and prompt about _confirming_ vehicle type (e.g. "Do you have a standard BlueBoat?" -> `Yes - Set Sail!` / `Nope - Custom Setup`), instead of _choosing_ it (from all options)? That way we could potentially avoid needing to create and maintain multiple separate BlueOS images for production (software's concern), but also avoid users receiving a stock vehicle with an identity crisis (SMS's concern).
>
> I think it seems reasonable for the base image to include the latest stable Sub and Rover firmware, in which case a default user could set up a standard vehicle completely offline (automatically loading the firmware for the relevant vehicle type), while custom users could range from setting up no autopilot, to installing a firmware for their autopilot, to also installing Extensions for any peripherals they've bought with the vehicle.

> @Williangalvani: That is a good idea - we should be able to identify the mikrotik radio and/or the fxti