bluerobotics/BlueOS #1829: core: frontend: multi-vehicle support: allow configuring system ID

ES-Alexander, 2023-06-28:
### Current behaviour

It's not possible to change nicely, because BlueOS services won't know about the change and may fail to connect to the autopilot

### Expected or desired behaviour

Allow setting the MAVLink system ID from the vehicle connection configuration dialog, as [previously planned](https://github.com/bluerobotics/BlueOS-docker/pull/1581#issuecomment-1500590624).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-06-28:
For now the user can do it via firmware parameter

joaoantoniocardoso, 2026-08-17:
Related: https://github.com/bluerobotics/BlueOS/issues/3604

patrickelectric, 2026-09-11:
Fixed in #4365, the system ID is configurable in Vehicle Setup.

closed 2026-09-11 by patrickelectric