bluerobotics/BlueOS #1830: bug: wizard progress bar has no section for internet

ES-Alexander, 2023-06-28:
### Bug description

When going through the wizard the first stage is internet connection, but there's no stage for it in the progress bar at the top, which means all subsequent stages show as one ahead in the progress compared to their actual state.

For example, this is the "Choose Vehicle" stage, but it shows as the Customise stage:

[image]

### Steps to reproduce

1. Run the wizard (e.g. on a fresh install of BlueOS, or by reducing the wizard version in Bag Editor and refreshing the page)
2. Check the progress bar at the top while going through the steps (they're all 1 ahead of where they're supposed to be)

### Primary pain point(s)

It's confusing to see it claiming to be in a different state to the actual one.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-06-28:
I saw the same behavior

joaoantoniocardoso, 2023-06-28:
There's the possibility of losing the internet after the internet step, so we probably should go back to that step if the connection fails.

joaoantoniocardoso, 2024-06-18:
I believe this is fixed:

[image]

closed 2024-06-18 by joaoantoniocardoso