bluerobotics/BlueOS #1855: [helper] Helper is eating CPU

joaoantoniocardoso, 2023-07-07:
It happens spontaneously when the system is being used, but it's easily replicable by reloading the BlueOS page ~10 times in a row:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/09303437-0f7c-4d15-a578-d4b519f3693e

Williangalvani, 2023-07-07:
yes, I suspect this is causing other issues, too. extension detection is one of them.
The caches should stop this from happening.

Maybe one of the services we probe is causing it?

joaoantoniocardoso, 2023-07-08:
> yes, I suspect this is causing other issues, too. extension detection is one of them. The caches should stop this from happening.
> 
> Maybe one of the services we probe is causing it?

Yeah, my test was wrong, it was a coincidence when I tried to load several times and it affected the service. What happens is that the cache does its job, but every 60 seconds it runs the service scanning loop, and this is what is demanding a lot of CPU. To replicate it, one just needs to wait and observe.

joaoantoniocardoso, 2023-07-08:
I'm writing a quick benchmark to see what can be done. I've tested updating the aiohttp to 3.8.4 and it helps a little, but it breaks the aiohttp-docker used by version-chooser. I have some small optimizations for the loops, we could switch requests to async aiohttp, but ultimately I think the scanning strategy/algorithm is the culprit, it waits 60 seconds and then uses all its power to quickly scan all as quickly as it can. Instead, it should use those 60 seconds to do that smoothly and quietly.

Williangalvani, 2023-07-08:
> I'm writing a quick benchmark to see what can be done. I've tested updating the aiohttp to 3.8.4 and it helps a little, but it breaks the aiohttp-docker used by version-chooser. I have some small [...]

I suspect one port in particular is to blame. Would you mind adding a debug log to show the current port being scanned? It is possible that just blacklisting it fixes the issue for now...

Williangalvani, 2023-07-08:
actually nevermind me. we already have that logged and for now, no particular service seems to be to blame.

I have two ideas, then:
 - stop re-scanning ports we already scanned, except for the docker auto-assigned ones (circa 327**)
 - improve the blacklist (i'd like to blacklist gdb's 5555 port)
 - check what the hell is triggering the chardet, which I suspect is pretty heavy
 - spreading the detection through the 60s as you mentioned, but that has some other implications... we'd need to always respond with a cached response, as we cant make the frontend wait 60s for the answer

joaoantoniocardoso, 2023-07-08:
> I suspect one port in particular is to blame. Would you mind adding a debug log to show the current port being scanned? It is possible that just blacklisting it fixes the issue for now...

[image]

Doesn't seem to be. Here I'm measuring time using time.process_time()

joaoantoniocardoso, 2023-07-08:
> * check what the hell is triggering the chardet, which I suspect is pretty heavy

The chardet is from aiohttp, but it's not exactly related to the port scanning time. I also disabled the logging as I thought it might be impacting, but didn't see a huge difference. 

Interestingly, if I disable the detect_service caching, it seems to lower the peaks.

joaoantoniocardoso, 2023-07-08:
> Interestingly, if I disable the detect_service caching, it seems to lower the peaks.

Nope, was measuring wrong. It helps a lot

joaoantoniocardoso, 2023-07-08:
Refactored to get rid of aiohttp and request, it's much lighter by directly using client.http:

https://github.com/bluerobotics/BlueOS-docker/assets/5920286/a7845c46-ce8a-43e4-812a-2732083ee27e

PR #1857 (2023-07-12): Optimize helper
## Changes

Most of the CPU was spent inside the HTTP requests done by `aiohttp` and `request`. By replacing them with the low-level `http.client`, both the speed and CPU usage were improved.

Closes #1855 

---

All development data and tools can be accessed [here](https://github.com/joaoantoniocardoso/BlueOS-docker/tree/new_helper_dev/core/services/helper/dev_benchmark)

## A/B test:
By doing several runs of the services, the port scanning is not idempotent, meaning that the list of available services can differ from one call to the other.

After analysis, it turns out that t [... cut]
closed 2023-07-12 by patrickelectric