bluerobotics/BlueOS #1903: How to create BlueOS-raspberry.zip file?

sakthivelj, 2023-07-27:


joaoantoniocardoso, 2023-07-27:
Hi, it's generated [on this CI step](https://github.com/bluerobotics/BlueOS/blob/1.1.0-beta.27/.github/workflows/test-and-deploy.yml#L260) using Pimod  :)

sakthivelj, 2023-10-15:
It is possible to generate BlueOS-raspberry.zip using GitHub action itself or VM raspberry pi?

patrickelectric, 2023-10-15:
Yes, as pointed in on @joaoantoniocardoso comments https://github.com/bluerobotics/BlueOS/issues/1903#issuecomment-1653494481, we do use github actions to generate it. But a real raspberry is necessary to run it in a [self hosted runner](https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/about-self-hosted-runners).