bluerobotics/BlueOS #1923: hotspot is always ON

joaoantoniocardoso, 2023-08-01:
>Hotspot is turned off, the icon in the wifi menu shows it is off, in the wifi >settings 'enable smart-hotspot' is turned off.
>
>Yet, uap0 is still up, and the network is visible on my other devices.

[image]
[image]
[image]

[image]

joaoantoniocardoso, 2023-08-01:
Possibly related to #1221

patrickelectric, 2025-01-09:
Should be fixed by https://github.com/bluerobotics/BlueOS/pull/2765

closed 2025-01-16 by patrickelectric