bluerobotics/BlueOS #1937: core: frontend: multi-language support

ES-Alexander, 2023-08-04:
### Current behaviour

Everything is in English. Many people are more comfortable with other languages, and could benefit from being able to view the interface in their preferred language.

This is particularly relevant because vehicle access is often without an internet connection, so it may not always be possible to use automatic (cloud-based) browser translation features.

### Expected or desired behaviour

Ideally users should be able to select their preferred language for the visual elements, and developers should be able to contribute translation files (perhaps even as a plugin / sub-extension, so they can be dynamically installed for people who want them).

A full implementation of this would likely require a significant refactor of the codebase, and I'm not sure how that can best be approached, or how it could be made to include support for extensions (they may need to be handled separately, although hopefully it will at least be possible to use a consistent system). That said, it's not critical to implement everything at once - it would likely be quite helpful to start with even just the button labels, and we could prioritise standard pages over pirate-mode only ones.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2026-11-06:
THAT WAS DONE 

[image]

Fixed by #4679

ES-Alexander, 2026-11-06:
> I don't think this should rely on extensions.

@rafaellehmkuhl Some benefits if we used [add-ons](https://github.com/bluerobotics/BlueOS/issues/1939) for this include:

1. Users could easily update their translations, without needing to update the software
2. Users could install only languages they want, instead of needing all of them
    - Having all of them would be pretty negligible in terms of storage, but would still making finding the translation you want a bit more annoying
3. Extensions could use the same interface, so they could be translated too

A benefit of having it built in is that you don't need to worry about compatible versions of translation files as different strings are introduced and/or updated in the source (but a good experience is then reliant on those translation files actually being kept updated as part of every release).