bluerobotics/BlueOS #1938: core: frontend: add theme/style configuration interface

ES-Alexander, 2023-08-04:
### Current behaviour

Theme style configuration is currently possible [with a css file](https://blueos.cloud/docs/blueos/1.1/advanced-usage/#theme-styling), but that's mostly useful for developers rather than general users. Easy styling is a "nice to have" for many users, but can also be an important accessibility feature for people with visual impairments.

### Expected or desired behaviour

It would be nice to have a pretty and intuitive UI for configuring the main page colours, ideally with a colour picker that shows the colours being selected (so it's not necessary to save a file and reload the page just to see the selected colour(s)). I think this is most appropriate as a built in service, but it could be developed as an extension if necessary / as a starting point.

It could also be nice to have a couple of selectable preset palettes that are tuned towards different types of colour-blindness, although ensuring the base theme is colour-blind friendly is a great starting point, and manual colour selection is a more valuable and powerful feature.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2024-07-05:
It may also be worth exploring custom theming via vuetify, rather than the current CSS overrides approach.

patrickelectric, 2026-09-11:
Fixed in #3930, the theme is configurable in Settings.

closed 2026-09-11 by patrickelectric