bluerobotics/BlueOS #1985: We should consider configuring Ping1D settings when starting up the Ping MAVLink driver

Williangalvani, 2023-08-12:
This actually needs investigation. If the settings are persistent and the user was playing with the ping before using it in a mission, we could have bad readings when using it with the driver.

ES-Alexander, 2023-08-15:
The BR Ping Sonar does not have persistent settings, so this only really applies if someone else creates a different device that follows the Ping1D protocol but does happen to have persistent settings.

Ping Viewer persists device settings configured through it, but I don't consider that relevant because if someone's connected to a sonar through Ping Viewer then they should expect that any settings they configure there will also be applied to the readings coming through the Ping1D MAVLink driver in BlueOS.

Williangalvani, 2023-08-15:
sounds like this can be closed, then.