bluerobotics/BlueOS #1987: [Commander] Ugly message to the frontend

joaoantoniocardoso, 2023-08-14:
[image]

RaulTrombin, 2023-08-14:
Same here. beta 28.
```
Could not send command to host: <html> <head><title>502 Bad Gateway</title></head> <body> <center><h1>502 Bad Gateway</h1></center> <hr><center>nginx/1.18.0</center> </body> </html> <!-- a padding to disable MSIE and Chrome friendly error page --> <!-- a padding to disable MSIE and Chrome friendly error page --> <!-- a padding to disable MSIE and Chrome friendly error page --> <!-- a padding to disable MSIE and Chrome friendly error page --> <!-- a padding to disable MSIE and Chrome friendly error page --> <!-- a padding to disable MSIE and Chrome friendly error page --> .
```

Williangalvani, 2023-08-14:
bad gateway? is commander dead? can you check the logs?

should we strip html away in the alerter? or just make sure they never get there?

RaulTrombin, 2023-08-14:
@Williangalvani logs from commander:
2023-08-14 16:43:16.717 | INFO     | __main__:<module>:34 - Starting Commander!
2023-08-14 16:43:23.228 | INFO     | __main__:setup_ssh:231 - SSH setup done
2023-08-14 16:43:23.230 | WARNING  | commonwealth.utils.commands:run_command:69 - Failed to run command with SSH key. , trying with sshpass:
ls

rotu, 2023-10-06:
This looks strange. Why does commander being dead result in a 502 error?

For context, I encountered the originally mentioned message when trying to set up `blueos-core` running on a MacBook, where I'm *sure* I have the docker setup not quite right.

> 2023-10-05 23:09:00 Running BlueOS: tag='1.1.1', commit_number='0', commit_hash='g2c9e558e'
> 2023-10-05 23:09:00 CompletedProcess(args=['sshpass', 'ssh', '-i', '/root/.config/.ssh/id_rsa', '-o', 'StrictHostKeyChecking=no', 'pi@localhost', 'uname -a'], returncode=255, stdout='', stderr='ssh: connect to host localhost port 22: Cannot assign requested address\n')
> 2023-10-05 23:09:00 Critical error: Something is wrong with the host computer, run_command is not working.
> 2023-10-05 23:09:00 Ignoring host computer configuration for now.
> 2023-10-05 23:09:00 Error: Docket socket file does not exist!

patrickelectric, 2023-10-06:
Hi @rotu.

We use nginx as a man in the middle for ous services to have a nice and clear path between the API and the microservices that we are running. So the 502 is nginx fault. 
https://github.com/bluerobotics/BlueOS/blob/e33f60d4091801b30a943f1d49489b8d35a94543/core/tools/nginx/nginx.conf#L86

PR #1992 (2023-08-15): core: commonwealth: fix ssh key path
oops
indirectly fixes #1987