bluerobotics/BlueOS #1991: no connectivity over ipv6

Williangalvani, 2023-08-15:
While this is likely an issue with my modem/ISP. it should be better handled.

for now, I did this:

`sudo nano /etc/sysctl.conf`

append the following lines to turn off ipv6:
```
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
run sudo sysctl -p to take effect or just reboot.
```

from https://raspberrypi.stackexchange.com/a/78278

joaoantoniocardoso, 2023-08-15:
Or just uncomment the `precedence ::ffff:0:0/96  100` line on the `/etc/gai.conf`

rotu, 2025-03-04:
Something resembling this seems to have been pushed out in https://github.com/bluerobotics/BlueOS/commit/715af4e8b0d3805a9b4b685417641410f0587a6a

Why disable IPv6? The system should have BOTH an IPv4 address and an IPv6 address if available.

If the issue was the Pi not getting and keeping an IPv4 address, I suspect the issue is actually a problem with the pi behaving badly on DHCP4: https://raspberrypi.stackexchange.com/questions/136320/raspberry-pi-loses-ipv4-address-randomly-but-keeps-ipv6-address

joaoantoniocardoso, 2025-03-04:
> Something resembling this seems to have been pushed out in [715af4e](https://github.com/bluerobotics/BlueOS/commit/715af4e8b0d3805a9b4b685417641410f0587a6a)
> 
> Why disable IPv6? The system should have BOTH an IPv4 address and an IPv6 address if available.
> 
> If the issue was the Pi not getting and keeping an IPv4 address, I suspect the issue is actually a problem with the pi behaving badly on DHCP4: https://raspberrypi.stackexchange.com/questions/136320/raspberry-pi-loses-ipv4-address-randomly-but-keeps-ipv6-address

Yeah, I believe we took the wrong approach there. We need support IPV6 for internet (maybe not necessarily for local networks). The problem is that ISPs frequently have broken IPV6 and it's not always easy to troubleshoot.

rotu, 2025-03-04:
> The problem is that ISPs frequently have broken IPV6 and it's not always easy to troubleshoot (e.g. it works today, it takes forever to load anything tomorrow, etc).

Okay. So *is* there a problem with IPv6 locally? I don't think there is - IPv6 has been more resilient locally than IPv4 in my experience.

If there's an issue with Internet/ISPs, I would think that the fix would be in DNS resolution settings, not in network interfaces.

joaoantoniocardoso, 2025-03-05:
> Okay. So _is_ there a problem with IPv6 locally? I don't think there is - IPv6 has been more resilient locally than IPv4 in my experience.
>
> If there's an issue with Internet/ISPs, I would think that [...]

Do you need IPv6 locally? If it's ipv6-only, I think the entire codebase should be checked, I believe most of the code was made with ipv4 locally, for example, mdns and vehicle's static.

rotu, 2025-03-05:
I mean the whole codebase should be checked regardless. There's some major weirdness going on with networking!