bluerobotics/BlueOS #2003: bug: [File-Manager] : User can't see new directory made from an extension

RaulTrombin, 2023-08-16:
### Bug description

File manager can't access new folders, red-pill + ls work and show the folder/files.

### Steps to reproduce

I'm trying create a new extension for navigator.
[Navigator Jupyter Extension](https://github.com/RaulTrombin/Navigator_PythonPlayground).

On the settings we have a persistant folder that binds to usr/blueos/extensions/.... (following the new standards we are discussing). Then user can save their own python code or anything else.

Docker command that create and keep the saved data:
```
    "Binds": [
      "/root/.config:/root/.config",
      "/usr/blueos/extensions/navigator_python_playground/notebooks:/work/notebook/persistent"
    ],
```

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2023-08-17:
apparently we need [an extra flag in our binds](https://docs.docker.com/engine/api/v1.43/#tag/Container/operation/ContainerCreate:~:text=%5B%5Br%5Dshared%7C%5Br,or%20slave.)

RaulTrombin, 2023-08-17:
On  [bond-core startup-json](https://github.com/bluerobotics/BlueOS/blob/master/bootstrap/startup.json.default
)
It's binding the blueos/userdata not the blueos :s

RaulTrombin, 2023-08-17:
[image]

suggest to use blueos folder instead.

Then we can use the standard:
/usr/blueos/extensions/<extension_name>/
Discussed here:
https://github.com/bluerobotics/BlueOS/issues/1640

PR #2010 (2023-09-07): Folder extensions
This PR able blueos-core to access /usr/blueos/extensions folder.

- Resolves #2003
- Enables the use of the standard discussed in #1640
closed 2023-09-07 by patrickelectric