bluerobotics/BlueOS #2042: bug: Log browser gives "folder fetch fail"

clydemcqueen, 2023-08-25:
### Bug description

Navigating to "Log browser" shows no logs, and gives the "folder fetch fail" error.

[image]

BlueOS Version: [tags/1.1.0-beta.28-0-g2df6ae03](https://github.com/bluerobotics/blueos-docker/releases/tag/1.1.0-beta.28)
Bootstrap Version: 1.0.1
Build: 8/12/2023, 8:51:38 PM
By [Blue Robotics](https://bluerobotics.com/)

Board name:
Pixhawk1
Manufacturer:
ArduPilot
Mavlink platform:
Pixhawk1
Firmware version:
4.1.0 (STABLE)
Vehicle type:
Submarine
Path:
/dev/ttyACM0

### Steps to reproduce

Click on Log browser.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2023-08-25:
Hi @clydemcqueen, thanks for reporting it.

I believe that folder is created once the vehicle is armed, in any case, we should create it beforehand to prevent that error message.

clydemcqueen, 2023-08-25:
I armed the vehicle using QGC, and I'm still seeing this behavior.

I'm running ArduSub on a Pixhawk on this system, so perhaps BlueOS "Log browser" will always be empty? I see 10 BIN files on the Pixhawk using QGC.

PR #2044 (2024-04-17): core: tools: logviewer: Create firmware logs folder beforehand
Fixes #2042
closed 2024-04-17 by patrickelectric