bluerobotics/BlueOS #2108: It would be cool to facilitate testing extensions

rafaellehmkuhl, 2023-09-14:
The main case is: an extension developer is developing some new features/bug-fixes, and he wants to test those changes or let someone test them. Those changes will likely be on a branch/pull-request on its GitHub repository.

The current setup for that is to enter its already-running extension container, clone the repo, checkout the branch and do what it needs to do there to make it run. It's very messy.

It would help the developers a lot if we could improve that.

@voorloopnul suggested a simple and effective solution: allow specifying a GitHub repository + branch name in the extension's manager.

From that, Kraken would clone the repo, build the Dockerfile there and run it.

It was mentioned that the building time could be long as it's being built on a Raspberry-like computer, but it's probably an acceptable downside as it makes both users and developer's life easier, and also because most extensions are probably going to be some Python scripts + a frontend, which will probably not take more time than what it takes today to apply an update on the BlueOS version.

voorloopnul, 2023-09-15:
[docs here](https://docs.docker.com/engine/reference/commandline/build/)

**build from a specific branch and context(folder)**
> docker build https://github.com/bluerobotics/BlueOS-examples.git#master:example3-vuetify

**build from default branch + root context**

> docker build https://github.com/Williangalvani/BlueOS-VirtualHere.git -t virtualhere

rafaellehmkuhl, 2023-09-15:
> [docs here](https://docs.docker.com/engine/reference/commandline/build/)
> 
> **build from a specific branch and context(folder)**
> 
> > docker build https://github.com/bluerobotics/BlueOS-examples.git#master:example3-vuetify
> 
> **build from default branch + root context**
> 
> > docker build https://github.com/Williangalvani/BlueOS-VirtualHere.git -t virtualhere
> 
> **Instead of branch it can also be a tag** _(fake example)_
> 
> > docker build https://github.com/Williangalvani/BlueOS-VirtualHere.git#valid_tag -t virtualhere

Holly hell. I didn't know this was actually built inside the docker CLI.

So we just need a frontend and a very small modification in Kraken, right?

ericjohnson97, 2024-02-25:
This might be need to be a seperate issue, but its mostly related. 

Today I fixed an issue with cockpit and the change was merged in. The CI generated a new container with tag master, but the extension UI first doesn't show the "master" tag so I was unable to install it via the UI. It would also be nice to be able to specify a custom url and tag to facilitate extension development. 

I was able to test the "master" tag by editing the start_extension method in kraken.py by adding 

```
extension.tag = "master"
``` 

To force kraken to use the "master" tag

It would be nice if there was an easier way to do this.

vshie, 2025-03-29:
my workflow has been to write code, publish to github, let the docker image built, and then install/update it, and test / check logs / etc. This is probably slower than building the image on the pi would be, I agree typically extensions are lightweight python flask/vue2 kinda things!