bluerobotics/BlueOS #2133: WiFi tray icon is showing `NetworkCard` from the System Info page

rafaellehmkuhl, 2023-09-22:
[image]

Williangalvani, 2023-09-26:
relevant (sporadic) error message:

```
[unplugin-vue-components] component "NetworkCard"(/home/will/git/blueos/blueos-docker/core/frontend/src/components/wifi/NetworkCard.vue) has naming conflicts with other components, ignored.
```

patrickelectric, 2023-09-27:
@rafaellehmkuhl can you inform how to replicate this issue ?
Master looks fine here

[image]

rafaellehmkuhl, 2023-09-27:
I dont know how to replicate the issue. I'm literally just opening the page.
Cache disabled.
Master image `718caeee`.

rafaellehmkuhl, 2023-09-27:
Updating to the latest master (`0c568f94`) didnt solve the issue.

@patrickelectric your screenshot is from production or local development?

PR #2141 (2023-09-27): core: frontend: components: wifi: Rename NetworkCard to WifiNetworkCard
Fix #2133
closed 2023-09-27 by joaoantoniocardoso