bluerobotics/BlueOS #2149: bug:

judahwo, 2023-10-03:
### Bug description

AUTO_VEHICLE_TYPE_FETCH_FAIL
[image]
I am on a Raspberry Pi 4B. It is not recognizing firmware that was previously installed just fine. Now it is throwing me an architecture incompatibility error
[image]
I basically can't do anything right now: 
[image]
The ethernet connection is fine:
[image]
Internet connection is fine: 
[image]
I am new to the device and the software, and can't start testing systems until this is resolved. This is for a school senior engineering design project. 

### Steps to reproduce

Repeated error every time I enter BlueOS through browser and IP

### Primary pain point(s)

I can't download vehicle firmware, I can't access the vehicle setup. I basically can't do much of anything except test my ethernet and internet connections. I'm still learning the interface, so not having basic functionality has me at a standstill. 

### Additional context

https://discuss.bluerobotics.com/t/autopilot-manager-vehicle-type-fetch-fail/15175

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

RaulTrombin, 2023-10-03:
Hi @judahwo, could you specify the version of BlueOS you are using? You can find this information in the bottom left corner of the interface.

Additionally, could you provide the output of the uname -a command? To access the system's terminal, you can navigate to the terminal tab, and then use the "red-pill" command, which will grant you access to the system's terminal, similar to an SSH connection.

[image]

judahwo, 2023-10-03:
Hello, @RaulTrombin thank you for your reply.

This is the version info at the bottom left:
[image]

And also in the terminal:
[image]

So, yes it is showing up as aarch64, as opposed to arm71. It's worth noting that this only started happening after the first time I accessed the 'red-pill' terminal and installed the Navigator Library (https://github.com/bluerobotics/navigator-lib). I tested some of the example functions found here: (https://docs.bluerobotics.com/navigator-lib/python/#), like the toggling of the onboard LED, PWM readings, etc. The next day I tried accessing BlueOS, and I had lost the MAVlink heartbeat and it no longer showed the firmware as installed. Basically, all the functionality that I had previously was gone and I am continuously getting the errors I described above.

RaulTrombin, 2023-10-03:
Nice to have you using the navigator libraries! 
You can also check the Jupyter extension, fast setup to start codding with python and run tests:

[image].
[image]
I'm using it on BlueOS 1.1.1, probably it's working fine with navigator libraries as it supports both armv7 and aarch64 architectures.

But, focusing on this issue:
Maybe an "apt get upgrade" command was made? 
It will force the raspberry to upgrade the kernel, and will needs some confirmation. As we can see your system is running on the 6.1.21-v8. And somehow the raspberry newer versions are forcing to boot on aarch64, not keeping the previous architecture ( which I think is wrong...it should keep the armv7, already opened an issue for this on rpi foundation). 

You can try to fallback the kernel to be the same that comes with official BlueOS images, it should fix it.

Or fresh install and use new stable BlueOS release, which have some great improvements: 
[BlueOS 1.1.1](https://github.com/bluerobotics/BlueOS/releases/tag/1.1.1)

judahwo, 2023-10-03:
I'm not sure about how fallback the kernel. I can try that. 

As for the fresh install, would that mean a new flash on my sd? I was using 1.1.1 stable originally:
[image]

Could the issue be that the bootstrap doesn't match?:
[image]

I see the option to update the bootstrap, but when I press it I get this warning:
[image]

So, I aborted that. Should I go ahead and update the bootstrap? Could that be my issue?

RaulTrombin, 2023-10-03:
the apt get update itself isn't a problem,
but if you run apt get `upgrade` it can be, as it updates the OS's kernel, and current BlueOS releases are based on the shown previous kernel.

`sudo SKIP_WARNING=1 rpi-update 1340be4`
should downgrade to the BlueOS current supported kernel

patrickelectric, 2023-10-03:
Hi @judahwo,

The problem that you are facing is a bit weird. Did you install BlueOS manually ? Are you using a BlueOS image that we provide ?

judahwo, 2023-10-03:
@patrickelectric , no, I used the SD card that came with the Navigator, with the image already on it. It worked correctly initially. I think that @RaulTrombin identified that a sudo apt update may have forced it into aarch64. I just bought and received the Navigator a couple of weeks ago, so I don't really know what I'm doing yet. It's quite possible that I changed something that I shouldn't have. I'm guessing it has something to do with me accessing the 'red pill' terminal doing that update.

judahwo, 2023-10-03:
@patrickelectric , now I don't know what's going on. I can't even pull up BlueOS with 192.168.2.2 in my browser. It's not working at all.

patrickelectric, 2023-10-03:
Hi @judahwo, 

From what I can see on your last reply here:
https://github.com/bluerobotics/BlueOS/issues/2149#issuecomment-1745506660

It appears that you kernel does not match with what comes with BlueOS image, any upgrade or update on the kernel will result in the following problem. 

Can you flash a new SD card with BlueOS ? 
Everything should work fine, when working with Navigator libraries, in C++, Python or Rust, a kernel update should not becessary.

I would recommend you installing and using the python libraries or using our own jupyter playground.

Let me know if you need any further assistance.

judahwo, 2023-10-03:
@patrickelectric yes, I am in class now, I was wondering if that is something I should try. I don't have a micro SD adapter with me, so I will follow the instructions in the software install guide and flash the SD card when I get home this evening. I will message here if I have any more issues. I won't do an update next time.

patrickelectric, 2023-10-03:
Okay, sorry for this issue. We need to make it more clear for the user that a kernel upgrade is dangerous.
Other than that, be free to ask for any further assistance.

patrickelectric, 2025-01-11:
This should be fixed in latest master

closed 2025-01-11 by patrickelectric