bluerobotics/BlueOS #2172: core: frontend:

CUIS0, 2023-10-17:
### Current behaviour

I noticed a bug in Blue OS where it replicates the ping 360 in the vehicle setup menu.

See the attached photo.

image
[image]

Do you have any additional context you can provide?

    is your Ping360 connected via USB or ethernet?Its connected using ethernet
    which BlueOS version are you using?using blue OS 1.1
    can you provide [system logs](https://blueos.cloud/docs/blueos/1.1/advanced-usage/#blueos-settings)?ill I’ll get it to you in a bit

questions from blue robotics:

    Are you able to consistently reproduce this issue? yes it’s persistent ever since the vehicle setup menu was modernized, even with a clean install of blueOS

    if so, can you provide the steps to do so? Turn on the ROV open vehicle setup and wait about an hour, then open the vehicle setup again

    Would you be willing to submit that information as a bug report in our [GitHub Issues](https://github.com/bluerobotics/BlueOS/issues)? I can try

### Expected or desired behaviour

Only one instance of ping360 is shown.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-10-17:
Hi @CUIS0, please provide the system log if possible.
@Williangalvani any idea ?

patrickelectric, 2024-02-19:
@CUIS0 please be free to open the issue if you find the necessary information for us to investigate or replicate the issue.

closed 2024-02-19 by patrickelectric