bluerobotics/BlueOS #2178: ardupilot-manager: support forcing firmware installs

ES-Alexander, 2023-10-26:
### Current behaviour

We have a tailored list of board types the manager supports, but that means new board types (or supported board types that are missed by our filters) need to have PRs made (e.g. #2177) before they can have firmware updates through BlueOS, which is frequently a slow process. Given many of those boards can actually be flashed by BlueOS if given an appropriate firmware, this seems like an unnecessary limitation on user behaviour.

The current workaround is to make a direct connection to QGC and flash there, but the validation fail warning doesn't say to do that, and doing it can be annoying (and requires having QGC installed).

### Expected or desired behaviour

We have a pirate mode. I think it should be possible/allowed for a pirate user to force download and install a firmware for a manually specified board type/name for boards that currently fail BlueOS's internal board type validation.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

lance-ljx, 2024-05-24:
@ES-Alexander How's that going now

ES-Alexander, 2024-05-24:
Hi @lance-ljx,

I'm not aware of any updates that have been made towards this, and I don't expect it will get done this quarter unless someone external to Blue Robotics implements and contributes the feature - you're welcome to give it a go if you're interested[^1] 🙂. Otherwise I'll try to raise this internally and see if we can get it added to next quarter's development goals.

It would likely help somewhat if #2551 gets implemented (because at least then it's less likely for a functioning flight controller board to be connected and not be recognised as a supported type), but that wouldn't fully resolve the issue because there could still be unidentified boards and/or a board with an existing identification that someone wants to swap the firmware for (e.g. between ArduPilot and PX4, or if a board has previously been flashed with incorrect firmware so it gets detected as the wrong type).

[^1]: I expect the main new component will be a pirate-mode-only platform selector that allows choosing between either all known platform types the firmware flasher supports, or ideally a filtered list of all platform types that may be compatible with the detected chip/hardware.