bluerobotics/BlueOS #2179: x86_64 Support

ericjohnson97, 2023-10-30:
### Current behaviour

I am interested in running Blue OS in an x86_64 VM to simulate fleets of systems. I am trying to run the ./install.sh script, but it returns

```
./install.sh 
Invalid architecture: x86_64
Supported architectures: armhf armv7 armv7l aarch64
```

### Expected or desired behaviour

It would be nice to support x86_64 as I could image robotic systems where a non ARM computer is chosen as the companion computer. 

Tangential to this, I would be interested in using the x86_64 vm as a development environment for improving and contributing to Blue OS, but based on the docs I have read I have not figured out a good way to set up a development environment.  Potentially adding a reference development environment to the docs would be a good improvement too. 

Thanks!

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2023-10-30:
I am not sure of quite what the situation is on X86 land right now. We do have amd64 images on dockerhub, so it _should_ be useable. You can try adding x86_64 to the script and following up any errors from there.

ericjohnson97, 2023-10-31:
Thanks for letting me know. I was able to get blueos bootstrap running tonight, but blueos core is crashing. I will continue debugging when I next get some time and report back!

patrickelectric, 2023-10-31:
Let us know the docker log to help you better

ericjohnson97, 2023-11-01:
```
eric@blueos:~$ docker logs blueos-core
bash: cannot set terminal process group (1): Inappropriate ioctl for device
bash: no job control in this shell
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: HWIDError { message: "No such file or directory (os error 2)", kind: "NotFound" }', src/main.rs:137:43
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

It looks like the core module is looking for a file or directory ... possibly configs.   

Is main.rs close source or in another repo? I can't seem to find it in the BlueOS repository.

on the bootstrap side it looks like it is trying the restart the core when it crashes, but thats probably not very interesting. 

```
2023-11-01 00:27:39.512 | INFO     | bootstrap.bootstrap:start:180 - Starting bluerobotics/blueos-core
2023-11-01 00:27:39.608 | INFO     | bootstrap.bootstrap:start:197 - core (bluerobotics/blueos-core:factory) started
2023-11-01 00:27:39.608 | WARNING  | bootstrap.bootstrap:run:245 - core is not running, starting..
2023-11-01 00:27:39.609 | WARNING  | bootstrap.bootstrap:is_version_chooser_online:222 - Could not talk to version chooser for 87.52481412887573: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /version-chooser/v1.0/version/current (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa9eff1d2e8>: Failed to establish a new connection: [Errno 111] Connection refused'))

```

patrickelectric, 2023-11-01:
Add the following arguments when running the docker:

`-mount type=bind,source=$PWD/blueos_root,target=/root/ -P -p 80:80 --pid=host --network=host`

ericjohnson97, 2023-11-01:
@patrickelectric Thanks for the hint. I tried that, and I am now able to get to the web page, but only some of the services are working. for instance, the setup wizard won't let me get past the configuration step. After looking at the network traffic, I think it is because there is no connection to the vehicle. I want to use a sitl instance as my vehicle so I tried running one, but no connection. I am not sure if the ardupilot vehicle manager also tries to auto autoconnect on udp 14550. I tried to track that down tonight, but I ran out of time.  

Also, is there a way to look at the logs from the services such as the ardupilot manger? I looked around, but couldn't find any outputted logs. Do I need to mount another volume to see service logs on the host? 

for reference. this is how I am now running blueos-core 

```
eric@blueos:~/BlueOS$ docker run --privileged --network=host --name=blueos-core   --mount type=bind,source=$PWD/.,target=/root/   -P -p 80:80 --pid=host   -v /var/run/docker.sock:/var/run/docker.sock   bluerobotics/blueos-core:factory
WARNING: Published ports are discarded when using host network mode
bash: cannot set terminal process group (205810): Inappropriate ioctl for device
bash: no job control in this shell
Error: blueos static system configuration folder does not exist!
Waring: Using dockers's resolv.conf, this may cause DNS issues if the host's network configuration changes.
Starting high priority services..
Service: autopilot: nice --19 /home/pi/services/ardupilot_manager/main.py
Service: cable_guy: /home/pi/services/cable_guy/main.py
Service: video: nice --19 mavlink-camera-manager --default-settings BlueROVUDP --mavlink tcpout:127.0.0.1:5777 --gst-feature-rank omxh264enc=0,v4l2h264enc=250,x264enc=260 --log-path /var/logs/blueos/services/mavlink-camera-manager --verbose
Service: mavlink2rest: mavlink2rest --connect=udpin:127.0.0.1:14000 --server 0.0.0.0:6040 --system-id 1 --component-id 194
Starting other services..
Service: beacon: /home/pi/services/beacon/main.py
Service: kraken: nice -19 /home/pi/services/kraken/main.py
Service: wifi: nice -19 /home/pi/services/wifi/main.py --socket wlan0
Service: bridget: nice -19 /home/pi/services/bridget/main.py
Service: commander: /home/pi/services/commander/main.py
Service: nmea_injector: nice -19 /home/pi/services/nmea_injector/nmea_injector/main.py
Service: helper: /home/pi/services/helper/main.py
Service: iperf3:  iperf3 --server --port 5201
Service: linux2rest: linux2rest
Service: filebrowser: nice -19 filebrowser --database /etc/filebrowser/filebrowser.db --baseurl /file-browser
Service: versionchooser: /home/pi/services/versionchooser/main.py
Service: pardal: nice -19 /home/pi/services/pardal/main.py
Service: ping: nice -19 /home/pi/services/ping/main.py
Service: user_terminal: cat /etc/motd
Service: ttyd: nice -19 ttyd -p 8088 sh -c "/usr/bin/tmux attach -t user_terminal || /usr/bin/tmux new -s user_terminal"
Service: nginx: nice -18 nginx -g "daemon off;" -c /home/pi/tools/nginx/nginx.conf
Service: log_zipper: nice -20 /home/pi/services/log_zipper/main.py '/shortcuts/system_logs/**/*.log' --max-age-minutes 60
Service: bag_of_holding: /home/pi/services/bag_of_holding/main.py
Service: major_tom: /home/pi/services/major_tom/main.py
BlueOS running!
```

ericjohnson97, 2023-11-02:
Tonight I figured out where the logs are stored. I noticed there was no folder for ardupilot_manager. looks like the ardupilot_manager service is crashing on launch with the following error.  

```
root@blueos:/home/pi/services/ardupilot_manager# nice --19 /home/pi/services/ardupilot_manager/main.py
Traceback (most recent call last):
  File "/home/pi/services/ardupilot_manager/main.py", line 24, in <module>
    from ArduPilotManager import ArduPilotManager
  File "/home/pi/services/ardupilot_manager/ArduPilotManager.py", line 21, in <module>
    from firmware.FirmwareManagement import FirmwareManager
  File "/home/pi/services/ardupilot_manager/firmware/FirmwareManagement.py", line 17, in <module>
    from firmware.FirmwareInstall import FirmwareInstaller
  File "/home/pi/services/ardupilot_manager/firmware/FirmwareInstall.py", line 9, in <module>
    from ardupilot_fw_decoder import BoardSubType, BoardType, Decoder
ModuleNotFoundError: No module named 'ardupilot_fw_decoder'
```

I could not find `ardupilot_fw_decoder` in the repo anywhere. any thoughts?

clydemcqueen, 2024-03-12:
`ardupilot_fw_decoder.py` is copied from ArduPilot's `firmware_version_decoder.py`. See https://github.com/bluerobotics/BlueOS/blob/3b40e44e7003e3ef17ee6de22debf496bb65eb69/core/tools/ardupilot_tools/bootstrap.sh#L41

closed 2025-12-10 by patrickelectric