bluerobotics/BlueOS #2185: move major tom to extension

patrickelectric, 2023-11-08:
### Current behaviour

Right now major tom is part of BlueOS, we should move it to its own extension.
Discussion: https://github.com/bluerobotics/BlueOS/pull/2183

### Expected or desired behaviour

With that major tom updates and release can be done outside BlueOS. 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2024-03-29:
- [ ] Update kraken to install default extensions automatically
   - [ ] We should check connection speed before install (for all extensions)
   - [ ] We should check tethered connection before install (for all extensions)
   - [ ] We should alert user if connection speed is slow or (tethered need to investigate) and ask for confirmation before proceeding
- [ ] Major tom should provide an status endpoint that helps kraken to verify if its ok
- [ ] Kraken should rollback to major tom factory version if major tom fails to start
- [ ] Kraken should be able to change major tom factory version and set a new default  

> Note we could put the available network interfaces on major tom if we desire to check for tethered network prefix

ES-Alexander, 2024-04-04:
- [ ] Presumably the default extensions should also come included with the pre-built RPi images?
- [ ] Also, I think it's important that Major Tom should provide an interface page showing what data is being collected (and potentially providing settings to disallow certain variables, or just a "turn off" button), and some stats on how many status updates have been sent, as well as how much data that's been.
    - It shouldn't be hard for users to get access to that information, and I expect telemetry is more palatable if it's very clear exactly what is being collected and sent.

patrickelectric, 2024-06-19:
Major tom is now blueos cloud agent