bluerobotics/BlueOS #2191: CI/CD: Enforce new line at the end of the file on pull requests

rafaellehmkuhl, 2023-11-09:
Right now the linter is not doing that, which causes the files to be modified to add or remove them, depending on the commit author.

Discussed [here](https://github.com/bluerobotics/BlueOS/pull/2188#discussion_r1388393138).

avelynhc, 2023-11-09:
I can work on adding [prettier](https://prettier.io/) to enforce a new line at EOF

patrickelectric, 2023-11-09:
> I can work on adding [prettier](https://prettier.io/) to enforce a new line at EOF

That would be really nice!

avelynhc, 2023-11-09:
Got it. I'll see what I can do with the existing black. Thanks!

rotu, 2023-11-09:
I don't think black and ruff support non-python files.

rotu, 2023-11-09:
And I think `black` and `ruff` already add it by default. And `shellcheck` which is currently in use doesn't format code (including that newline at end of line).

I think prettier might be the best bet for non-python files in this repo.

patrickelectric, 2023-11-09:
An alternative would be to use https://github.com/mvdan/sh shfmt

rotu, 2023-11-12:
@patrickelectric you’re right. Prettier doesn’t support shell scripts, and the shell script plugin for prettier uses shfmt internally: https://github.com/un-ts/prettier/tree/master/packages/sh