bluerobotics/BlueOS #2205: bug: Freshly installed BlueOS: 502 Gateway Unavailable errors

dieter-aerit, 2023-11-30:
### Bug description

I just installed BlueOS from scratch, and something seems broken.
[image]

### Steps to reproduce

Install BlueOS

### Primary pain point(s)

It doesn't work out of the box, looks bad.

### Additional context

v1.1.1

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-11-30:
Hi @dieter-aerit! I'm sorry that was your initial experience, how did you install it and in which board ?
Also, if you have access to the terminal, please [install neofetch](https://github.com/dylanaraps/neofetch/wiki/Installation) and the output of the command.

If you are using a raspberry pi 4 and used the install script, I would recommend using our raspberry image: https://github.com/bluerobotics/BlueOS/releases/tag/1.1.1
For more information about installing it, check: https://blueos.cloud/docs/blueos/1.1/

dieter-aerit, 2023-11-30:
Hi @patrickelectric ! Just watching your YouTube talk about BlueOS as I'm typing this :).

I clicked the link here that gives me the latest release from GitHub:
[image]

Then I flashed that image onto an SD card using the RaspberryPi imager.
Then inserted the SD into a RaspberryPi 4, waited for it to boot and visited the BlueOS interface.

The Wizard did not work for me, it got blocked on trying to fetch a list of available parameter sets?
So then I refreshed the UI and skipped the wizard, after which I basically saw the screenshot included in the bug report. No further idea of what is causing this.

Neofetch output:
[image]

patrickelectric, 2023-11-30:
Thanks!

So.. What is probably happening is that there is no flight controller connected, pixhawk or navigator on your raspberry. By default I believe that is not running SITL, to do that.

Go to blueos -> AutoPilot Firmware -> Change Board -> SITL
That will solve some of the issues related to missing mavlink data.

The video stream issue is more confuse, do you have a camera connected ? Is the video stream service running at all ? You can check that enabling pirate mode under the happy robot icon on the top right corner, it'll enable more options. From there, you can go on available services page on the page selection menu. You should see a "MAVLink Camera Manager" service there on the list, running on port 6020.

patrickelectric, 2023-11-30:
@joaoantoniocardoso any idea why the camera manager is giving such error ?

patrickelectric, 2023-11-30:
@dieter-aerit can you provide the logs of your system ?

[image]

patrickelectric, 2025-01-09:
@dieter-aerit is it working ?