bluerobotics/BlueOS #2207: Websockets don't work reliably on dev server since the migration to Vite

Williangalvani, 2023-12-04:


joaoantoniocardoso, 2023-12-04:
example..?

Williangalvani, 2023-12-05:
on both my machines:

mac m1 with node v18.12.1 and Arch linux with Node 20.3.1

`[yarn|bun] dev `  results in websocket taking more than 3 minutes to connect.
adding `--host` seems to fix the issue. I'll investigate further.

Williangalvani, 2023-12-14:
Reopening. The issue is is reproducible if yarn is running on a different computer than the one you are browsing from.

patrickelectric, 2024-01-14:
is this still an issue ? what are you experiencing with it ?

Williangalvani, 2024-01-23:
yeah. it is happening right now. I am running `BLUEOS_ADDRESS=http://192.168.15.132 yarn dev --host` on a machine, and accessing the frontend from a different machine.
All websockets are pending (except one used by the dev server for hotreload):
[image]

patrickelectric, 2024-01-23:
did you try this ?
https://stackoverflow.com/questions/69481783/http-requests-stuck-in-pending-state-when-using-vite/69673499#69673499

Williangalvani, 2024-02-01:
> did you try this ? https://stackoverflow.com/questions/69481783/http-requests-stuck-in-pending-state-when-using-vite/69673499#69673499

That did solve the issue. I'm very confused. why are there so many open files?

Well. I guess I'll close the issue, but maybe we should document this somewhere?

Williangalvani, 2024-02-01:
Also documented here:
https://vitejs.dev/guide/troubleshooting.html#dev-server

closed 2024-02-01 by Williangalvani