bluerobotics/BlueOS #2210: Kraken is kind of dying sometimes. Appears to be something related to accessing extension logs.

rafaellehmkuhl, 2023-12-05:
This happens when trying to access logs from an extension. I tried with Cockpit initially, but others gave the same.

[image]

[image]

joaoantoniocardoso, 2023-12-05:
What version are you running? Is this related to https://github.com/bluerobotics/BlueOS/pull/2124 ?

rafaellehmkuhl, 2023-12-05:
> What version are you running? Is this related to #2124 ?

I don't think so.

```
BlueOS Version: [heads/master-0-ge85c728a](https://github.com/bluerobotics/blueos-docker/tree/e85c728a)
Bootstrap Version: 1.1.1
Build: 11/28/2023, 5:54:06 PM
By [Blue Robotics](https://bluerobotics.com/)
```

Williangalvani, 2023-12-05:
@rafaellehmkuhl can you try williangalvani/blueos-core:stream_logs ?

rafaellehmkuhl, 2023-12-05:
> @rafaellehmkuhl can you try williangalvani/blueos-core:stream_logs ?

Tested. No luck.

Williangalvani, 2023-12-05:
I can't replicate. what extensions do you have installed?

rafaellehmkuhl, 2023-12-06:
Those 3:
[image]

Still happening on latest master.

Attaching the logs below:

[_kraken.zip](https://github.com/bluerobotics/BlueOS/files/13574221/_kraken.zip)

Williangalvani, 2023-12-06:
after some debugging, it seems like the vscode extension docker is.... insanely huge.
[logfile_12-06-2023_02_38_16.log](https://github.com/bluerobotics/BlueOS/files/13575237/logfile_12-06-2023_02_38_16.log)

is it trying to use pretty much all the ports in the system? 65k udp and 65k tcp:
```
>>> 131001-122
130879
>>> _/2
65439.5
```

Williangalvani, 2023-12-06:
@patrickelectric any ideas of what could be causing this? I couldn't find anything obvious on the dockerfile or anywhere else. Docker's documentation barely mentions "PrivatePort", too.

patrickelectric, 2023-12-06:
The ports are exposed via a range argument: https://github.com/BlueOS-community/blueos-code-server/blob/master/Dockerfile#L48

Williangalvani, 2023-12-06:
I wonder if there is a way to do this dynamically. or maybe we should just do net=host? 

the root of the issue here is that the json returned by Docker's API and resulting python object are huge due to the 130k port entries.

I only see two ways out, we either increase the memory usage limit for Kraken or change this extension

patrickelectric, 2023-12-06:
Can't we just filter our the Ports property from the API if it goes over a range ? Like: 100 ports ?
I believe that using ports range will be pretty common.

Williangalvani, 2023-12-06:
we would have to reimplement the aiodocker `containers.list()` function, which isn't ideal. plus we'd be getting the huge json in memotry anyway to get rid of the 130k entries.

Do you recall why you had to add this? I don't understand how it works, since the allocated port won't be the port that vscode expects it to be

patrickelectric, 2023-12-06:
VSCode extension, as jupyter, aims to be a development platform. 
For that, it's necessary to have a range of ports that the user can use to access BlueOS services and at the same time user services.
To do that, I have created a range 100-MAX for the ports, that's not for vscode, but for the user code/software that is running on the docker.

patrickelectric, 2024-07-09:
I believe that we never had this problem again, closing it until further information

closed 2024-07-09 by patrickelectric