bluerobotics/BlueOS #2218: Services should be automatically restarted when killed due to high memory usage, and the user should be notified that this happened

Williangalvani, 2023-12-06:


Williangalvani, 2023-12-07:
I see we doing that long-term. but we need a short-term fix.
We could have a wrapper around the services that runs then in a while true loop, delaying 5s after they die, or something similar

joaoantoniocardoso, 2023-12-07:
@Williangalvani 

edit: the spawned process from the terminal should not interrupt the terminal, so a while loop should indeed just work :)

Williangalvani, 2023-12-07:
our issue is not even the OOM killer, we introduced python code into our services to limit their memory usage, some of them are handling worse than others

joaoantoniocardoso, 2023-12-07:
> our issue is not even the OOM killer, we introduced python code into our services to limit their memory usage, some of them are handling worse than others

OOOh, I forgot about that. So the "MemoryError" from #2232 is that memory limitation, probably?

Williangalvani, 2023-12-07:
> OOOh, I forgot about that. So the "MemoryError" from #2232 is that memory limitation, probably?

yes

PR #2250 (2023-12-12): Core: update start_blueos_core to add memory limit and logging
(opening as draft because I have issues with shellcheck on my machine)

fix #2218
closed 2023-12-12 by patrickelectric