bluerobotics/BlueOS #2225: We need to be able to run extension images that are not released...

rafaellehmkuhl, 2023-12-07:
...just like we do for BlueOS.

BlueOS repos are usually configured to generate and publish images of the PRs, and BlueOS can fetch and run those images, making testing PRs easy.

We need to replicate this mechanism to the extensions. Be it by fetching images from the docker registry or allowing sideloading.

Williangalvani, 2023-12-08:
we already do?

all you need to do is write a different tag here:
[image]

Williangalvani, 2023-12-08:
> I think it's not the ideal final scenario yet

I'm listening =]

rafaellehmkuhl, 2023-12-08:
> > I think it's not the ideal final scenario yet
> 
> I'm listening =]

I think the most straightforward for the final user would be to only have to select a tag from a dropdown menu, as it does with the releases right now. We could have it under a separated dropdown that is only shown under pirate mode.

rafaellehmkuhl, 2023-12-08:
What I'm saying is that we should have a one step process. It should be easy for the user if it can be.

rafaellehmkuhl, 2023-12-09:
Thanks @Williangalvani! The suggestion worked like a charm!

@ES-Alexander this is for sure something we should have documented!

rafaellehmkuhl, 2023-12-09:
May I add that today when you try to click "create", it will blame you for not adding custom settings in the last text input. I think it should just use the default ones in this case.