bluerobotics/BlueOS #2231: [Helper] Internet speed is broken

joaoantoniocardoso, 2023-12-07:
### Decription

Just open the internet speed test page, the service fails with: `pydantic.error_wrappers.ValidationError: 11 validation errors for SpeedTestResult`

BlueOS Version: heads/master-0-gff96e5a3 Bootstrap Version: master Build: 12/7/2023, 3:53:11 PM By Blue Robotics 

Relevant log part:

```Python
2023-12-07 16:47:36.613 | ERROR    | commonwealth.utils.apis:custom_route_handler:36 - Unhandled service exception.                                                                                                                            
2023-12-07 16:47:36.641 | ERROR    | commonwealth.utils.apis:custom_route_handler:37 - 11 validation errors for SpeedTestResult                                                                                                                
response -> server -> url                                                                                                                                                                                                                      
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> lat                                                                                                                                                                                                                      
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> lon                                                                                                                                                                                                                      
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> name                                                                                                                                                                                                                     
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> country                                                                                                                                                                                                                  
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> cc                                                                                                                                                                                                                       
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> sponsor                                                                                                                                                                                                                  
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> id                                                                                                                                                                                                                       
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> host                                                                                                                                                                                                                     
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> d                                                                                                                                                                                                                        
  field required (type=value_error.missing)                                                                                                                                                                                                    
response -> server -> latency                                                                                                                                                                                                                  
  field required (type=value_error.missing)                                                                                                                                                                                                    
Traceback (most recent call last):                                                                                                                                                                                                             
                                                                                                                                                                                                                                               
  File "/home/pi/services/helper/main.py", line 561, in <module>                                                                                                                                                                               
    loop.run_until_complete(server.serve())                                                                                                                                                                                                    
    │    │                  │      └ <function Server.serve at 0xb5902808>                                                                                                                                                                     
    │    │                  └ <uvicorn.server.Server object at 0xb58c8310>                                                                                                                                                                     
    │    └ <function BaseEventLoop.run_until_complete at 0xb63c9b20>                                                                                                                                                                           
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>                                                                                                                                                                           
                                                                                                                                                                                                                                               
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 634, in run_until_complete                                                                                                                                                      
[... cut ...]
response -> server -> sponsor
  field required (type=value_error.missing)
response -> server -> id
  field required (type=value_error.missing)
response -> server -> host
  field required (type=value_error.missing)
response -> server -> d
  field required (type=value_error.missing)
response -> server -> latency
  field required (type=value_error.missing)
2023-12-07 16:47:36.861 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:38482 - "GET /latest/internet_test_previous_result HTTP/1.0" 500
```

Williangalvani, 2023-12-08:
it is also throwing a lot of  `RuntimeError: can't start new thread` which I'm unable to get around...
I even changed all the code back to async, only to find that the error still raises from within the speedtest library.

threading.active_count() never showed more than 10 threads.

joaoantoniocardoso, 2023-12-08:
> it is also throwing a lot of `RuntimeError: can't start new thread` which I'm unable to get around... I even changed all the code back to async, only to find that the error still raises from within the speedtest library.
> 
> threading.active_count() never showed more than 10 threads.

I thought that that thread error was related to another part of the code :eyes:

PR #2257 (2023-12-13): core: helper: bump memory use limit to 200mb
fix #2231
closed 2023-12-13 by rafaellehmkuhl