bluerobotics/BlueOS #2233: [Autopilot firmware] Sometimes fetching firmware returns nothing

joaoantoniocardoso, 2023-12-07:
I failed to get the console logs, but at least I have an image:

[image]

BlueOS Version: heads/master-0-gff96e5a3 Bootstrap Version: master Build: 12/7/2023, 3:53:11 PM By Blue Robotics

joaoantoniocardoso, 2023-12-20:
Maybe related:

[image]

joaoantoniocardoso, 2024-01-11:
On master it's now happening every time:

```Python
> File "/home/pi/libs/commonwealth/commonwealth/utils/apis.py", line 31, in custom_route_handler
    return await original_route_handler(request)                                                                                                                                                                                                                            [115/2773]
                 │                      └ <starlette.requests.Request object at 0xb4048ab0>                                                                                                                                                                                           
                 └ <function get_request_handler.<locals>.app at 0xb52d3d98>                                                                                                                                                                                                          
                                                                                                                                                                                                                                                                                      
  File "/usr/local/lib/python3.11/site-packages/fastapi/routing.py", line 274, in app                                                                                                                                                                                                 
    raw_response = await run_endpoint_function(                                                                                                                                                                                                                                       
                         └ <function run_endpoint_function at 0xb5c4d578>                                                                                                                                                                                                             
  File "/usr/local/lib/python3.11/site-packages/fastapi/routing.py", line 193, in run_endpoint_function                                                                                                                                                                               
    return await run_in_threadpool(dependant.call, **values)                                                                                                                                                                                                                          
                 │                 │         │       └ {'vehicle': <Vehicle.Sub: 'Sub'>, 'board_name': 'Navigator'}                                                                                                                                                                   
                 │                 │         └ <function get_available_firmwares at 0xb52d3c08>                                                                                                                                                                                       
                 │                 └ <fastapi.dependencies.models.Dependant object at 0xb532a4d0>                                                                                                                                                                                     
                 └ <function run_in_threadpool at 0xb5e1d6b8>                                                                                                                                                                                                                         
  File "/usr/local/lib/python3.11/site-packages/starlette/concurrency.py", line 41, in run_in_threadpool                                                                                                                                                                              
    return await anyio.to_thread.run_sync(func, *args)                                                                                                                                                                                                                                
                 │     │         │        │      └ ()                                                                                                                                                                                                                                 
                 │     │         │        └ functools.partial(<function get_available_firmwares at 0xb52d3c08>, vehicle=<Vehicle.Sub: 'Sub'>, board_name='Navigator')                                                                                                                 
                 │     │         └ <function run_sync at 0xb5e2eb68>                                                                                                                                                                                                                  
                 │     └ <module 'anyio.to_thread' from '/usr/local/lib/python3.11/site-packages/anyio/to_thread.py'>                                                                                                                                                                 
[... cut ...]
                   └ <zlib.Decompress object at 0xb402d968>                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                      
MemoryError: Unable to allocate output buffer.
```

joaoantoniocardoso, 2024-01-11:
@patrickelectric the same is happening on beta 1.20.0;

rafaellehmkuhl, 2024-01-11:
I'm also facing this on master.

joaoantoniocardoso, 2024-01-11:
It seems to be memory-related. Probably our memory limit patch.

joaoantoniocardoso, 2024-01-11:
Correcting: it doesn't *always* happen, I restarted and it vanished on both versions. It may need to be triggered by event. If it is memory-related, we might have a memory leak that increases the baseline, then, the request process is just enough to pass that threshold.

joaoantoniocardoso, 2024-01-15:
I'll keep watching to check if #2299 has fixed this problem.

joaoantoniocardoso, 2024-01-31:
Reopening because it is still happening:

[image]

patrickelectric, 2024-03-06:
@joaoantoniocardoso are you still having the problem ?

joaoantoniocardoso, 2024-03-06:
@patrickelectric just tested multiple times, couldn't reproduce.

closed 2024-03-06 by joaoantoniocardoso