bluerobotics/BlueOS #2248: BLUE OS reverts to factory after shutdown

nimapefra, 2023-12-10:
### Bug description

After shutting down using the Ip address (blues Frontend, the next day, or randomly, when I restart, it reverts to factory.
So I have to reconfigure blue os again.
Simple as that, as well as annoying, and this makes the whole thing unsuable

### Steps to reproduce

1 setup autopilot, and then firmware
2 shutdown .
wait untile message says : you can power off.
3 restart..maybe after few hours. 
The system has recovered to factory.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

nimapefra, 2023-12-10:
[image]

patrickelectric, 2023-12-10:
Hi @nimapefra, Thanks for reporting the issue.
If you can, please apply the version 1.1.1 of BlueOS and be sure that bootstrap is updated to this version as well, if you are in pirate mode, a button called "Update Bootstrap" should show up.
You can check bootstrap version on the left side bar on the bottom.
We have a couple of improvements on bootstrap to avoid such behavior. 
If you can, please provide your system log for us to investigate further.

patrickelectric, 2023-12-10:
Possible duplication of #1600

joaoantoniocardoso, 2023-12-10:
> Possible duplication of #1600

The 1600 seems related to memory when uploading a new BlueOS, so unlikely to be the same issue.

nimapefra, 2023-12-10:
Hello, I have followed your steps exactly.After that I have commanded power-off from the blue OS before shutting down.After receiving the message that I can shutdown, I turned the ROV offAfter repowering, it has recovered to factoryAnd I have to re upload frmware and OS versionIs there a battery issue?Is there a battery that holds the information for the next reboot?Let me know at the moment it resets after every power down/And, additionally, I have a continuous annoying -mygcs255 heartbeatlost-- with no reason.Connection looks good, as well as powering option

nimapefra, 2023-12-10:
I am using latest Qground Control,in conjuction with NavigatorThanks

nimapefra, 2023-12-10:
I have instead tried to flash the 1.1.2 beta, and the relative bootstrap...I have turned it off and on a a couple of times...it seems to hold.Since it's late night I will try again tomorrowBut still I have this annoying ''Heartbit lost '' message every on eminute or 2 that is driving me crazystill I havent plugged any thruster or light or sonar...I am just building the ROVAnd basically only a Ublox Gps has been isntalled so far, along with a couple of 5v fan to dissipate heat

nimapefra, 2023-12-12:
Hello, I still have problems booting the BlueOS. Hvae tried with al the stable Versions and firmwares, including relative bootstrap
The next day, or hour, after being shutdown, it will reset to factory setting. That makes the navigator pretty unusable, isnt it?What do I have to do next?

nimapefra, 2023-12-12:
Hi Costas -
The vehicle reverting to factory is a bug our team is chasing at the
moment. It is worth noting that the fix is at least a quick one - the 1.1.1
firmware remains installed, and does not require the internet to apply. The
battery holds no information and does not communicate.
As far as the heartbeat lost message, I've seen this can be connected to
your client computer - if it is running a video call, and so under load, or
is not a very powerful machine, this message will occur more frequently.
Thanks for your patience!

patrickelectric, 2023-12-12:
Hi @nimapefra,

Sorry! I missed somehow the response for this post, I thought that was already replied.
So, based in what you said. I highly recommend to check to latest stable (1.1), go to settings, clear the log, and try to replicate the issue.
If it goes to factory again, check settings and download the log to send it to us.
It's necessary to analyze such logs to identify the problem.

joaoantoniocardoso, 2023-12-25:
Closed by #2265

closed 2023-12-25 by joaoantoniocardoso

nimapefra, 2023-12-31:
Hi , I have read the messages.I am not sure the issue of reverting to factory has found a solution.Do I have to manual enter the bootstrap Script and manually change the Time setting to monotonic?Or do I have to do something with the Version Chooser?Thanks. See you in 2024 :)

 
 

Closed #2248 as completed.

patrickelectric, 2023-12-31:
Hi @nimapefra, right now we don't have a beta or stable release with the fix. We should work on one next year.
But, if you want to apply right now:

- go to version chooser
- check master version
- update bootstrap to master as well
- go to the BlueOS version that you wish after bootstrap update

This will ensure that bootstrap is updated to master while you can use any BlueOS version

nimapefra, 2024-01-01:
Thanks for your info. I have noticed that the Navigator ServoRail (where all thrusters and servos) , by thecnical specification, should require 5v input to receive power. , as described here......
[cid:12c84301-2861-4f88-b01b-49c349c1644b]
But I see that servo rail works fine even without plugging any aux 5v power supply input.
Let me know what I am not getting. thanks again, happy new year
Costa

nimapefra, 2024-02-18:
Hello, good morning. Regarding the issue of BLUE os reverting to factory, after attempting randomly different firmware and versions, the problem stopped from arising. Which is good.
Please support me regarding this last issue I am experiening.
For unknown reasons, while I use the rov in the pool for some tests,  I notice after a while that the Escs of my 6 thrusters sound the alarm, once every 3 seconds. I am using this 40 amp ESC type...I know they might  not be the best association, but they where working great with the Pixhawk /Ardusub combination

More in detail:This issue can happen even with charged battery ( 16.8v fully charged), when I demand Gain to 100% to the thruster, and /or I intensively use the ROV devices at once, like lights, scanner, sonar.
Additionally, If /when I use the robotic arm, when the arm swings in position (its a significant rotation effort ), from vertical to horizontal position, the effort of the servo causes also (and every signle  time)  a momentary hiccup of the entire navigator, and  a momentary loss of data/control. All this never happened using the Pixhawk controller, where I had installed all the same equipment , but only 3 thrusters,smaller.
I suspect that the instant amp draw, using intensive devices, like 6 thrusters at high gain, or swining Robotic arm in position, causes a momentary hiccup on the Navigator board, which causes in return a warning alarm to all the ESC, even at fully charged status. (like when they turn on the car engine, and the stereo/hifi shutsdown for an instant, due to the instant ampdraw).

I have assured that the Navigator servo rail is getting the AUX 5v input. (robotic arm and thrusters signal are attached to the servo rail).

Is it possible that the navigator suffers from having 6 thrusters connected on the servo rail? The P75 thrusters are correctly operating in their 14v range. I have also removed the robotic arm.The hiccup/esc warning alarm shows up still, at high thrusters demand.The battery is the same identical   that they use on the BlueRov , ordered from same supplier (18000mah, 14v (16.8V fully charged))
I know that is hard to give any answer, and it can be a wide array of reasons.I have searched the forum, I dont think I have found similar issues, so probably I have made some mistakes, or asking too much from the navigator board.Do I need maybe a sort of  ZeroCrossing Switch (ZCS) circuit. to limit the surge current ? I have never needed that before actually.

 I have a custom ROV prototype, that uses your Navigator, a raspberry PI4 standard,the same battery 18000mah 14V that you commonly use.A cerluean Imaging sonar, a PIng2 echosounder.cerulean ROV locator (12v)Enginges are P75 thrusters, 12v each40 amps esc.
Thanks for your time and support.Best regardsCosta
Message ID: ***@***.***>

nimapefra, 2024-02-20:
Hi Costa -
BlueOS v1.2 beta 2 fixed the revert bug, this fix will be present in the
upcoming stable 1.2 release!

As for your issue with your ESCs and Thrusters, I'm unable to support
troubleshooting a piece of hardware that Blue Robotics doesn't sell! From
the description of your problem, the voltage supplied to the system is
dropping too low during high-current use. You could increase the size of
your wires between the batteries and the ESCs, or add a second battery in
parallel.
If your robotic arm is using the same 5V power supply that is connected to
the Navigator you need more power capacity than the 3A it is providing for
your load, given the "brown-out" behavior. Adding a second 5V DCDC power
supply to power your arm is likely the best solution.
The ESCs do not pull any power from the Navigator servo rail, only receive
a signal. A ZCS circuit is not relevant to this DC power device.
I'm not familiar with the P75 thruster, but they could pull more power than
a T200 - making larger wires necessary. Best of luck!

On Sun, Feb 18, 2024 at 5:38 AM 'giuseppe Costantino' via Support <
***@***.***> wrote:

> Hello, good morning. Regarding the issue of BLUE os reverting to factory,
> after attempting randomly different firmware and versions, the problem
> stopped from arising. Which is good.
>
> Please support [...]

-- 

*Tony White*Applications Engineer
Blue Robotics

Visit us virtually:
Blue Robotics <http://bluerobotics.com/> | Instagram
<https://www.instagram.com/bluerobotics/> | Forums
<http://discuss.bluerobotics.com/> | Product Feedback/Request Form
<https://app.smartsheet.com/b/form/9dce4ad233d14bc3a0ee60e79c224e87>