bluerobotics/BlueOS #2298: bug: Known good firmware file flashed via file upload, and not automatic web download, results in non-operable vehicle

vshie, 2024-01-13:
### Bug description

Flashed new 1.2.0 beta.1 onto SD card when old Pi 4 died, found that flashing ardusub from file doesn't work. 

### Steps to reproduce

Running 1.2.0 beta.1
downloading from here: 
https://firmware.ardupilot.org/Sub/stable/navigator/
Autopilot menu> Flash from file
Flashed OK
In cockpit or QGC, arm vehicle, no response from paired joystick / no motor movement. 
Disarm vehicle - hear initialization tones from ESCs. 
Change of MOT_SAFE_DISARM has no effect on this behavior. 
Reverted to factor ardusub, 4.1.0, behavior of ESCs / motors nominal. 

### Primary pain point(s)

This may be preventing other people with custom firmware from uploading and having it work, and would prevent those in the field without internet from changing versions. 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2024-01-14:
#2300 and #2299 may fix the issue

patrickelectric, 2024-02-19:
@vshie is this still an issue ?

vshie, 2024-02-19:
I flashed from latest beta without this issue happening, so I think it is fixed!

patrickelectric, 2024-02-19:
Thanks @vshie!

closed 2024-02-19 by patrickelectric