bluerobotics/BlueOS #2306: autopilot: Request timed out after 1.0 second.

joaoantoniocardoso, 2024-01-15:
```python
commonwealth.mavlink_comm.exceptions.MavlinkMessageReceiveFail: Request timed out after 1.0 second.                                                                                                     
2024-01-15 16:24:10.574 | INFO     | logging:callHandlers:1706 - 127.0.0.1:52544 - "GET /v1.0/vehicle_type HTTP/1.0" 500                                                                                
2024-01-15 16:24:10.576 | ERROR    | commonwealth.utils.apis:custom_route_handler:36 - Unhandled service exception.                                                                                     
2024-01-15 16:24:10.577 | ERROR    | commonwealth.utils.apis:custom_route_handler:37 - Request timed out after 1.0 second.                                                                              
Traceback (most recent call last):                                                                                                                                                                      
                                                                                                                                                                                                        
  File "/home/pi/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py", line 139, in send_mavlink_message                                                                                         
    async with session.post(                                                                                                                                                                            
               │       └ <function ClientSession.post at 0xb55062a8>                                                                                                                                    
               └ <aiohttp.client.ClientSession object at 0xb3e6f030>                                                                                                                                    
                                                                                                                                                                                                        
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 1117, in __aenter__                                                                                                            
    self._resp = await self._coro                                                                                                                                                                       
    │    │             │    └ <member '_coro' of '_BaseRequestContextManager' objects>                                                                                                                  
    │    │             └ <aiohttp.client._RequestContextManager object at 0xb5266c38>                                                                                                                   
    │    └ <member '_resp' of '_BaseRequestContextManager' objects>                                                                                                                                     
    └ <aiohttp.client._RequestContextManager object at 0xb5266c38>                                                                                                                                      
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 448, in _request                                                                                                               
    with timer:                                                                                                                                                                                         
         └ <aiohttp.helpers.TimerContext object at 0xb3e6f170>                                                                                                                                          
  File "/usr/local/lib/python3.11/site-packages/aiohttp/helpers.py", line 656, in __exit__                                                                                                              
    raise asyncio.TimeoutError from None                                                                                                                                                                
          │       └ <class 'TimeoutError'>                                                                                                                                                              
          └ <module 'asyncio' from '/usr/local/lib/python3.11/asyncio/__init__.py'>                                                                                                                     
                                                                                                                                                                                                        
TimeoutError                                                                                                                                                                                            
                                                                                                                                                                                                        
                                                                                                                                                                                                        
The above exception was the direct cause of the following exception:

                                                                                                                                                                                              [740/1452]
Traceback (most recent call last):                                                                                                                                                                      
                                                                                                                                                                                                        
  File "/home/pi/services/ardupilot_manager/main.py", line 292, in <module>                                                                                                                             
    loop.run_until_complete(server.serve())                                                                                                                                                             
    │    │                  │      └ <function Server.serve at 0xb5978a78>                                                                                                                              
    │    │                  └ <uvicorn.server.Server object at 0xb52a0d70>                                                                                                                              
    │    └ <function BaseEventLoop.run_until_complete at 0xb626c2a8>                                                                                                                                    
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>                                                                                                                                    
                                                                                                                                                                                                        
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 640, in run_until_complete                                                                                                              
    self.run_forever()                                                                                                                                                                                  
    │    └ <function BaseEventLoop.run_forever at 0xb626c258>                                                                                                                                           
[... cut ...]
          │    └ <commonwealth.mavlink_comm.MavlinkComm.MavlinkMessenger object at 0xb52a1bf0>
          └ <commonwealth.mavlink_comm.VehicleManager.VehicleManager object at 0xb5390f70>

  File "/home/pi/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py", line 146, in send_mavlink_message
    raise MavlinkMessageSendFail(f"Request timed out after {request_timeout} second.") from error
          └ <class 'commonwealth.mavlink_comm.exceptions.MavlinkMessageSendFail'>

commonwealth.mavlink_comm.exceptions.MavlinkMessageSendFail: Request timed out after 1.0 second.
```

patrickelectric, 2024-02-19:
is this still a bug ?

joaoantoniocardoso, 2024-02-19:
Not replicable anymore.

closed 2024-02-19 by joaoantoniocardoso