bluerobotics/BlueOS #2309: [network test] Failed to do speed test: Request failed with status code 502

joaoantoniocardoso, 2024-01-15:
Maybe related to #2308.

[image]

On the backend (Pardal):

```python
2024-01-15 22:43:58.875 | INFO     | logging:callHandlers:1706 - 127.0.0.1 [15/Jan/2024:22:43:50 +0000] "GET /get_file HTTP/1.0" 200 104857758 "http://localhost:8080/tools/network-test" "Mozilla/5.0 (X11; Linux x86_64; rv:121.0) Gecko/20100101 Firefox/121.0"
2024-01-15 22:44:13.522 | ERROR    | logging:callHandlers:1706 - Fatal error: protocol.data_received() call failed.
protocol: <RequestHandler connected>
transport: <_SelectorSocketTransport fd=11 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):

  File "/home/pi/services/pardal/main.py", line 88, in <module>
    web.run_app(app, path="0.0.0.0", port=args.port)
    │   │       │                         │    └ 9120
    │   │       │                         └ Namespace(port=9120)
    │   │       └ <Application 0xb5a401d0>
    │   └ <function run_app at 0xb5afe938>
    └ <module 'aiohttp.web' from '/usr/local/lib/python3.11/site-packages/aiohttp/web.py'>

  File "/usr/local/lib/python3.11/site-packages/aiohttp/web.py", line 508, in run_app
    loop.run_until_complete(main_task)
    │    │                  └ <Task pending name='Task-1' coro=<_run_app() running at /usr/local/lib/python3.11/site-packages/aiohttp/web.py:429> wait_for=...
    │    └ <function BaseEventLoop.run_until_complete at 0xb62bcf28>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 640, in run_until_complete
    self.run_forever()
    │    └ <function BaseEventLoop.run_forever at 0xb62bced8>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 607, in run_forever
    self._run_once()
    │    └ <function BaseEventLoop._run_once at 0xb62bee38>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1922, in _run_once
    handle._run()
    │      └ <function Handle._run at 0xb63253e8>
    └ <Handle _SelectorSocketTransport._read_ready()>
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
    │    │            │    │           │    └ <member '_args' of 'Handle' objects>
    │    │            │    │           └ <Handle _SelectorSocketTransport._read_ready()>
    │    │            │    └ <member '_callback' of 'Handle' objects>
    │    │            └ <Handle _SelectorSocketTransport._read_ready()>
    │    └ <member '_context' of 'Handle' objects>
    └ <Handle _SelectorSocketTransport._read_ready()>
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 952, in _read_ready
    self._read_ready_cb()
    │    └ <bound method _SelectorSocketTransport._read_ready__data_received of <_SelectorSocketTransport fd=11 read=polling write=<idle...
    └ <_SelectorSocketTransport fd=11 read=polling write=<idle, bufsize=0>>
> File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1009, in _read_ready__data_received
    self._protocol.data_received(data)
    │    │         │             └ b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0...
    │    │         └ <function RequestHandler.data_received at 0xb5b717a8>
    │    └ <RequestHandler connected>
    └ <_SelectorSocketTransport fd=11 read=polling write=<idle, bufsize=0>>
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 348, in data_received
    self._message_tail += data
    │    │                └ b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0...
    │    └ <member '_message_tail' of 'RequestHandler' objects>
    └ <RequestHandler connected>

MemoryError
```

patrickelectric, 2024-02-19:
@joaoantoniocardoso is this still an issue ?

joaoantoniocardoso, 2024-02-19:
Not replicable anymore.

closed 2024-02-19 by joaoantoniocardoso