bluerobotics/BlueOS #2326: We should delete older extensions once update is done

patrickelectric, 2024-01-20:
The system is not being cleaned

ES-Alexander, 2024-01-26:
We definitely need a mechanism for deleting old extension images, but from a "you own what you've paid for" perspective I think users should be given a choice whether they want to (since they may not be able to download them again if the developer has removed the version from hosting), instead of it being forcibly automatic, and they should also be able to roll back to old versions at will (like we allow for the core image with VersionChooser).

Williangalvani, 2024-02-01:
I guess the issue is here:

https://github.com/bluerobotics/BlueOS/blob/master/core/services/kraken/kraken.py#L197-L210

we should do remove(..., True) in order to delete. but we have to do it only after we confirm the pull was successful

patrickelectric, 2024-02-19:
Related to #2365

PR #2641 (2024-06-04): Kraken: Add extension Module V2
Depends on #2640

Closes #2326
closed 2024-06-04 by patrickelectric