bluerobotics/BlueOS #2329: core: frontend:

dgudiel, 2024-01-23:
### Current behaviour

The option to start/run the configuration wizard currently uses the word "enable". Using a different word, like "start" or "run", would better convey that pressing the button will cause the wizard to start immediately. 

### Expected or desired behaviour

Use a different word, like "Start Vehicle Configuration Wizard" or "Run Vehicle Configuration Wizard", to better convey that pressing the button will cause the wizard to start immediately.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-01-23:
@dgudiel I like the _"Run Vehicle Configuration Wizard"_ option.

There is also a tooltip when you leave the mouse over the button. Today it says _"Enable Wizard"_, any suggestions for it?

dgudiel, 2024-01-23:
I suppose it should match whatever the button is changed to, so either "Run Wizard" or "Start Wizard".

PR #2334 (2024-01-23): core: frontned: Change text from 'Enable Wizard' button
Closes #2329
closed 2024-01-23 by patrickelectric