bluerobotics/BlueOS #2369: vehicle setup: lights configuration showing for rover vehicle

patrickelectric, 2024-02-06:
There is also another thing, the page end up showing for rover as well, should we remove it ?

_Originally posted by @patrickelectric in https://github.com/bluerobotics/BlueOS/issues/2368#issuecomment-1930747586_

ES-Alexander, 2024-02-06:
I'm pretty sure rover doesn't have joystick-based lights control, but we could maybe replace it with controls for the NavLight (including perhaps some nice configuration options of the flashing pattern, for example)?

Williangalvani, 2024-02-06:
yeah, I'm thinking where we should filter it. Maybe filter here?
https://github.com/bluerobotics/BlueOS/blob/master/core/frontend/src/components/vehiclesetup/Configure.vue#L54

PR #2371 (2024-02-07): core: frontend: vehicle setup: allow filtering setup tabs
fix #2369
closed 2024-02-07 by patrickelectric