bluerobotics/BlueOS #2388: scrolling in tmux is too fast

Williangalvani, 2024-02-15:
makes it very annoying to read text. At least on my touchpad.

could be fixed by setting `set -g @scroll-speed-num-lines-per-scroll 5` to a lower value than default. not sure what default is

patrickelectric, 2024-02-19:
Weirdly, it's fine here for my devices

ES-Alexander, 2024-02-29:
I agree with @Williangalvani - scrolling on other sites is fine, but BlueOS terminal is quicker than ideal.

This is me moving pretty much the smallest amount that I can:

https://github.com/bluerobotics/BlueOS/assets/25898329/bb2c3d81-63b8-4bf1-94d1-910fe440d904

@patrickelectric are your devices all set super zoomed out? If there are lots of lines displayed then presumably moving several lines at once is less of an issue because you can still easily get the part you want at least somewhere on the screen.

closed 2026-08-31 by joaoantoniocardoso