bluerobotics/BlueOS #2396: MCM won't work if vehicle ID is different from 1

rafaellehmkuhl, 2024-02-19:


joaoantoniocardoso, 2024-02-19:
Could you elaborate?

rafaellehmkuhl, 2024-02-19:
If the user deploys BlueOS on a vehicle and changes the Ardupilot ID to a number other than 1, this won't be passed to the MCM command line.

This is purely a theory, looking at the code and discussing with the rest of the team thought. I didn't test it.

joaoantoniocardoso, 2024-02-19:
Right. MCM is now relying on the environment variable, which cannot be dynamically changed.

rafaellehmkuhl, 2024-02-19:
Perfect. So it's a bug indeed.

I've opened it because there are some people already trying blueos+cockpit with a fleet of vehicles, and so we are adding support for it on Cockpit, but it would only work for telemetry as of now.

joaoantoniocardoso, 2024-02-19:
BTW, if the person uses Cockpit, the Mavlink ID for the MCM would be irrelevant.

rafaellehmkuhl, 2024-02-19:
> BTW, if the person uses Cockpit, the Mavlink ID for the MCM would be irrelevant.

Humnnnn, true!
We don't check it.
It would only be relevant for QGC for example, that relies on the ID, right?